PixlOne/logiops #1: using bluetooth

iamzubin, 2019-07-26:
hey, 
so I've been using this with my MX Master 2S
and works pretty well, but when using it with Bluetooth, hidpp doesn't find the device(tried using the tools in their repo)
However, I can find the device manually.
can you please add an optional tag so we can find the device path and put it manually

PixlOne, 2019-07-27:
I don't actually have a bluetooth 4.0 adapter atm so I can't really test it but the device path is /dev/hidrawX, right?

Try removing the device name check between line 33 and 51 on src/logid/DeviceFinder.cpp and tell me if that fixes anything.

iamzubin, 2019-07-28:
yes, the device is /dev/hidraw4 but removing the name check doesn't help as the hidpp doesn't recognise it as a hidpp2.0 device even though when it sends any input, its identical to the data received on a USB connection.

I tried tweaking the library, but couldn't find any way to use my custom path

PixlOne, 2019-08-30:
Sorry for a bit of a late response but I've been looking at this on my Raspberry Pi that supports Bluetooth 4.0 and it doesn't seem to recognize my MX Master as an HID++ device. It seems that the device is using the hid-generic module instead of the hid-logitech-hidpp module. This seems to be an issue with the kernel's hid-logitech-hidpp module (at least on 4.19). I'm going to see if I can possibly patch the kernel to fix this.

PixlOne, 2019-09-12:
This bug has been fixed in my latest commit to hidpp: https://github.com/PixlOne/hidpp/commit/96b13e95ef18e514f72652a8e0bd99e1f2927991

closed 2019-09-12 by PixlOne