PixlOne/logiops #3: Implement human readable Ids instead of just Command Ids in config files

rockerbacon, 2019-09-11:
I was thinking of adding mappings from human readable button names to their respective CIds so that configuration will be more comfortable.

My thinking on how this could be done is:
1. _DeviceFinder_ calls a _DeviceFactory_ instead of directly instantiating a device. _DeviceFactory_ is responsible for associating a device name to the correct map. If the device does not have defined mappings it is instantiated without one;
2. _Device_ uses the map during the processing of the configuration file. If the device does not have a mapping it rejects human readable names, accepting only CIds (as it currently is).

I'm glad to work on this myself, but I'd like some input first (and the CIds for the MX Master 2S, if possible).

In the meantime it's also worth to add to the documentation that CIds can be found with the command:
```
hidpp20-reprog-controls --device <device index> <device path> info
```
Where _device index_ and _device path_ can be found with the command:
```
hidpp-list-devices
```

PixlOne, 2019-09-12:
Yeah, the CID idea was a temporary "hack" designed to make configuration quick to implement. In the final product, logid.cfg isn't meant to be edited directly; instead, logid.cfg is supposed to be auto-generated by a program. I'm planning on writing a GUI configuration utility which has human-readable diagrams of the buttons which are mapped to CIDs.

rockerbacon, 2019-09-12:
What I like about having this on the back-end instead of the front-end is that it keeps the information centralized and independent. In the Linux world we have GTK, KDE, Pantheon, etc, and it would be nice for any developer wanting to build a GUI to be able to use a unified system instead of redoing the same thing again.

I'll get some code going, even if it isn't used in this project it'll be useful for the GUI you plan on working on.

PixlOne, 2019-09-12:
Well the GUI is going to interface with logid so anyone can write a GUI implementation that interfaces with it. I could also write a map with CID nicknames and CIDs. In the meantime, I'll probably drop a table of CIDs & their functions in my wiki.

PixlOne, 2019-09-12:
I wrote a table of CID functions over here: https://github.com/PixlOne/logiops/wiki/CIDs