PixlOne/logiops #7: Unable to install

ThanosDi, 2019-10-22:
Hello I'm trying to make and install this repo but I'm unable.

<details><summary>Below are the steps:</summary>

```
auser@PowerPC:~/P/logiops|master$ mkdir build
auser@PowerPC:~/P/logiops|master$ cd build/
auser@PowerPC:~/P/l/build|master$ cmake ..
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.5.8
-- Checking for module 'systemd'
--   Found systemd, version 237
-- Checking for module 'libconfig'
--   Found libconfig, version 1.5
Could not find libhidpp include dir, getting submodule
-- Checking for module 'libudev'
--   Found libudev, version 237
-- systemd units will be installed at /lib/systemd/system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/auser/Programs/logiops/build
auser@PowerPC:~/P/l/build|master$ make
Scanning dependencies of target hidpp
[  1%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/misc/Log.cpp.o
[  2%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/misc/CRC.cpp.o
[  4%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/RawDevice_linux.cpp.o
[  5%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/DeviceMonitor_linux.cpp.o
[  7%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/UsageStrings.cpp.o
[  8%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/Dispatcher.cpp.o
[ 10%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/SimpleDispatcher.cpp.o
[ 11%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/DispatcherThread.cpp.o
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In constructor ‘HIDPP::DispatcherThread::DispatcherThread(const char*)’:
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:70:8: warning: suggest explicit braces to avoid ambiguous ‘else’ [-Wdangling-else]
     if (!checkReportDescriptor (rdesc))
        ^
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In member function ‘virtual void HIDPP::DispatcherThread::sendCommandWithoutResponse(HIDPP::Report&)’:
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:108:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
     if (forceLongReports() && report.type() == Report::Short)
     ^~
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:110:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  _dev.writeReport (report.rawReport ());
  ^~~~
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In member function ‘virtual std::unique_ptr<HIDPP::Dispatcher::AsyncReport> HIDPP::DispatcherThread::sendCommand(HIDPP::Report&&)’:
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:115:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
     if (forceLongReports() && report.type() == Report::Short)
     ^~
[... 101 lines cut ...]
/home/auser/Programs/logiops/src/logid/hidpp/src/libhidpp/hidpp20/MacroFormat.cpp:214:71: warning: unused parameter ‘device’ [-Wunused-parameter]
 std::unique_ptr<AbstractMacroFormat> HIDPP20::getMacroFormat (Device *device)
                                                                       ^~~~~~
[ 88%] Linking CXX shared library libhidpp.so
[ 88%] Built target hidpp
Scanning dependencies of target logid
[ 89%] Building CXX object src/logid/CMakeFiles/logid.dir/logid.cpp.o
In file included from /home/auser/Programs/logiops/src/logid/Device.h:6:0,
                 from /home/auser/Programs/logiops/src/logid/Actions.h:3,
                 from /home/auser/Programs/logiops/src/logid/util.h:4,
                 from /home/auser/Programs/logiops/src/logid/logid.cpp:12:
/home/auser/Programs/logiops/src/logid/Configuration.h:5:10: fatal error: libconfig.h++: No such file or directory
 #include <libconfig.h++>
          ^~~~~~~~~~~~~~~
compilation terminated.
src/logid/CMakeFiles/logid.dir/build.make:62: recipe for target 'src/logid/CMakeFiles/logid.dir/logid.cpp.o' failed
make[2]: *** [src/logid/CMakeFiles/logid.dir/logid.cpp.o] Error 1
CMakeFiles/Makefile2:87: recipe for target 'src/logid/CMakeFiles/logid.dir/all' failed
make[1]: *** [src/logid/CMakeFiles/logid.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
</details>

Any assistance you can provide would be greatly appreciated..

PixlOne, 2019-10-24:
I'm not quite sure what issue you're having, can you post some logs?

m-j-r, 2019-10-30:
I'm getting the exact same problem, here is the failing part of my ```make``` command.

```
canning dependencies of target logid
[ 89%] Building CXX object src/logid/CMakeFiles/logid.dir/logid.cpp.o
In file included from logiops/src/logid/Device.h:6,
                 from logiops/src/logid/Actions.h:3,
                 from logiops/src/logid/util.h:4,
                 from logiops/src/logid/logid.cpp:12:
logiops/src/logid/Configuration.h:5:10: fatal error: libconfig.h++: No such file or directory
    5 | #include <libconfig.h++>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/logid/CMakeFiles/logid.dir/build.make:63: src/logid/CMakeFiles/logid.dir/logid.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:96: src/logid/CMakeFiles/logid.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

PixlOne, 2019-10-31:
I'm sorry, I didn't realise that the `Below are the steps` text was expandable to show the log and I just realised that now.

Although, you were missing libconfig, CMake should have warned you beforehand so while it is technically user error, there is still a bug within the CMakeLists.

closed 2019-10-31 by ThanosDi

jgoodman8, 2020-01-26:
@ThanosDi how did you solve? I got the same issue and I have libconfig installed.

`
jg@pc:~/logiops/build$ ldconfig -p | grep libconfig
libconfig.so.9 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libconfig.so.9
libconfig.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libconfig.so
`

PixlOne, 2020-01-28:
Is it possible that you're missing the headers for libconfig? Depending on your distro, you may need a package called `libconfig-dev` (Ubuntu) or `libconfig-devel` (Fedora/RHEL).

jgoodman8, 2020-01-28:
Yes, I had a look and I think the .h files are missing. However, `libconfig-dev` it's already installed.

PixlOne, 2020-01-30:
It seems that you would need to install the libconfig++-dev package.

jgoodman8, 2020-01-30:
Great! That was the problem. I didn't know that both libraries were installed separately. Thanks!

Nanotorre, 2020-09-21:
Hi. I' m stuck at the same errror. 

```
-- Checking for module 'libevdev'
--   No package 'libevdev' found
```
And I have it installed with the libconfig++-dev package. Any idea?

```
~/logiops$ ldconfig -p | grep libconfig
	libconfig.so.9 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libconfig.so.9
	libconfig.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libconfig.so
	libconfig++.so.9 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libconfig++.so.9
	libconfig++.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libconfig++.so
```
```
dpkg --search libconfig
libconfig-dev:amd64: /usr/share/info/libconfig.info.gz
libconfig9:amd64: /usr/share/doc/libconfig9
libconfig-doc: /usr/share/doc/libconfig-doc/copyright
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/examples/example1.c
libconfig-doc: /usr/share/doc/libconfig-doc
libconfig9:amd64: /usr/lib/x86_64-linux-gnu/libconfig.so.9.2.0
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples/example.cfg
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples/example1.cpp
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/examples/example2.c
libconfig9:amd64: /usr/share/doc/libconfig9/README
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/copyright
libconfig++9v5:amd64: /usr/lib/x86_64-linux-gnu/libconfig++.so.9.2.0
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/examples/example3.c
libconfig9:amd64: /usr/lib/x86_64-linux-gnu/libconfig.so.9
libconfig9:amd64: /usr/share/doc/libconfig9/TODO
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/examples
libconfig-doc: /usr/share/doc/libconfig-doc/libconfig.pdf.gz
libconfig++9v5:amd64: /usr/share/doc/libconfig++9v5
libconfig++9v5:amd64: /usr/lib/x86_64-linux-gnu/libconfig++.so.9
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/changelog.Debian.gz
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/changelog.Debian.gz
libconfig++9v5:amd64: /usr/share/doc/libconfig++9v5/changelog.Debian.gz
libconfig++-dev:amd64: /usr/lib/x86_64-linux-gnu/libconfig++.a
libconfig++9v5:amd64: /usr/share/doc/libconfig++9v5/copyright
libconfig-dev:amd64: /usr/share/doc/libconfig-dev
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples/example2.cpp
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/examples/debian.Makefile
libconfig-doc: /usr/share/doc/libconfig-doc/changelog.Debian.gz
libconfig-dev:amd64: /usr/lib/x86_64-linux-gnu/libconfig.so
libconfig-doc: /usr/share/doc-base/libconfig9
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples/invalid.cfg
libconfig++-dev:amd64: /usr/include/libconfig.h++
libconfig-dev:amd64: /usr/lib/x86_64-linux-gnu/libconfig.a
libconfig-dev:amd64: /usr/lib/x86_64-linux-gnu/pkgconfig/libconfig.pc
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples/debian.Makefile
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev
libconfig-dev:amd64: /usr/include/libconfig.h
libconfig++-dev:amd64: /usr/lib/x86_64-linux-gnu/libconfig++.so
libconfig9:amd64: /usr/share/doc/libconfig9/copyright
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples/example4.cpp
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/copyright
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples/example3.cpp
libconfig-dev:amd64: /usr/share/doc/libconfig-dev/examples/example.cfg
libconfig++-dev:amd64: /usr/lib/x86_64-linux-gnu/pkgconfig/libconfig++.pc
libconfig++-dev:amd64: /usr/share/doc/libconfig++-dev/examples
libconfig9:amd64: /usr/share/doc/libconfig9/changelog.Debian.gz
```

PixlOne, 2020-10-02:
You need the `libevdev-dev` package.