PixlOne/logiops #9: Can't get gestures to work

m-j-r, 2019-10-31:
Great project - much appreciated, thanks.

I'm having trouble getting gestures to work - please help.

Here are the steps I've taken so far:

1. Using the example config, the gesture button doesn't work
2. If I change the example config to have a Keypress action instead of gesture for the same cid (0x52) then the key is pressed
3. If I change the example config to have a Keypress action for direction: "None", then that key is pressed, but no other direction works
4. If I remove the direction: "None" entry, then gestures don't work

When pressing the gesture button, the mouse pointer doesn't move - as if there is a gesture happening, but nothing happens.

I'm using an MX Master, on Kernel 5.3.7.

Happy to spend some time debugging - let me know how I can help.

PixlOne, 2019-11-01:
That's strange, I'm using the MX Master on Arch Linux, kernel 5.3.7 too and my gesture button works fine. The mouse movements and press/release events are clearly being diverted too.

You could try peeking around into [logid/Actions.cpp](https://github.com/PixlOne/logiops/blob/master/src/logid/Actions.cpp) and debugging `GestureAction::Move()` and `GestureAction::Release()`.

If you can figure this out, please share how you fixed this. If not, then could you share some more info that could help me reproduce this bug? Thanks.

PixlOne, 2019-11-01:
Also, here's the configuration file I regularly use. See if this also has that bug:
```
devices: (
    {
        name: "MX Master";
        smartshift:
        {
            on: true;
            threshold: 30;
        };
        hiresscroll:
        {
            hires: true;
            invert: false;
            target: false;
        };
        dpi: 1000;

        buttons: (
            {
                cid: 0xc3;
                action =
                {
                    type: "Gestures";
                    gestures: (
                        {
                            direction: "Up";
                            mode: "Axis";
                            axis: "REL_WHEEL";
                            axis_multiplier: 0.6;
                        },
                        {
                            direction: "Down";
                            mode: "Axis";
                            axis: "REL_WHEEL";
                            axis_multiplier: -0.6;
                        },
                        {
                            direction: "Left";
                            mode: "OnRelease";
                            action =
                            {
                                type: "Keypress";
                                keys: ["BTN_SIDE"];
                            };
                        },
                        {
                            direction: "Right";
                            mode: "OnRelease";
                            action =
                            {
                                type: "Keypress";
                                keys: ["BTN_EXTRA"];
                            };
                        },
                        {
                            direction: "None"
                            mode: "OnRelease"
                            action =
                            {
                                type: "Keypress";
                                keys: ["BTN_MIDDLE"];
                            };
                        }
                    );
                };
            },
            {
                cid: 0xc4;
                action =
                {
                    type: "Keypress";
                    keys: ["KEY_MSDOS"];
                };
            },
            {
                cid: 0x53;
                action =
                {
                    type: "Keypress";
                    keys: ["KEY_UNKNOWN"];
                }
            }
        );
    }
);
```

comio, 2019-11-01:
@PixlOne thanks for your working configuration. Is there any way to keep the "ALT+TAB" behaviour pressing (and keeping pressed) the 0xc3 button?

PixlOne, 2019-11-01:
Does my configuration work? On my setup, both the example config and my config work fine, and not to mention the format seems to be the same.

m-j-r, 2019-11-04:
Hi, I tried your config, and the same issue occurs...

I have tried debugging a bit on my machine, and without actually understanding your code (so these may be totally off the mark), the following things I find suspicious:

1. In [Device.cpp > ButtonHandler::handleEvent()](https://github.com/PixlOne/logiops/blob/38dcc65d4b2692f2bd0bb5f39141ca57e54ce2ac/src/logid/Device.cpp#L282) event.function() is always DivertedButtonEvent, never DivertedRawXYEvent which I think should be the case for a gesture?
2. [Actions.cpp > GestureAction::move()](https://github.com/PixlOne/logiops/blob/38dcc65d4b2692f2bd0bb5f39141ca57e54ce2ac/src/logid/Actions.cpp#L82) is never called, meaning that in [Actions.cpp > GestureAction::release()](https://github.com/PixlOne/logiops/blob/38dcc65d4b2692f2bd0bb5f39141ca57e54ce2ac/src/logid/Actions.cpp#L126) the x and y values are always 0, meaning that Direction is always taken to be None.

Let me know how I should proceed - I understand that I am taking stabs in the dark, so let me know if I'm totally off the track.

Thanks for producing these libraries, and helping me to debug my setup.

PixlOne, 2019-11-06:
`event.function() is always DivertedButtonEvent, never DivertedRawXYEvent which I think should be the case for a gesture?`

DivertedButtonEvent is called when a diverted button is pressed/released; DivertedRawXYEvent is called when the mouse is moved while a button with diverted mouse movements is held. 

It's really strange that DivertedRawXYEvent is never called. Can you post your distribution, mouse firmware info, and a list of your device features using [hidpp-list-features from here](https://github.com/PixlOne/hidpp)?

m-j-r, 2019-11-12:
Hi there, so sorry it took me so long to get back on this.

Distro is [clear linux](https://clearlinux.org/)
Mouse firmware...? - Don't know how to find this out.

Here is the output from hidpp-list-features, and following is some random commands from the same repo
<pre><font color="#4E9A06"><b>❯</b></font> sudo ./hidpp-list-features /dev/hidraw1
MX Master (046d:b012) is a HID++ 4.5 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x1d4b] Wireless device status
Feature 0x05: [0x0020] Reset
Feature 0x06: [0x1000] Battery status
Feature 0x07: [0x1814] Change host
Feature 0x08: [0x1b04] Reprog controls v4
Feature 0x09: [0x2201] Adjustable dpi
Feature 0x0a: [0x2100] Vertical scrolling
Feature 0x0b: [0x2110] Smart shift
Feature 0x0c: [0x2121] Hi-res wheel
Feature 0x0d: [0x6501] Gesture 2
Feature 0x0e: [0x1813] ? (hidden, internal)
Feature 0x0f: [0x1830] ? (hidden, internal)
Feature 0x10: [0x1890] ? (hidden, internal)
Feature 0x11: [0x18a1] ? (hidden, internal)
Feature 0x12: [0x18c0] ? (hidden, internal)
Feature 0x13: [0x1df3] ? (hidden, internal)
Feature 0x14: [0x1e00] Enable hidden features (hidden)
Feature 0x15: [0x1eb0] ? (hidden, internal)
Feature 0x16: [0x1803] ? (hidden, internal)
Feature 0x17: [0x1861] ? (hidden, internal)
Feature 0x18: [0x9000] ? (hidden, internal)
Feature 0x19: [0x9200] ? (hidden, internal)
Feature 0x1a: [0x9240] ? (hidden, internal)
Feature 0x1b: [0x1805] OOBState (hidden, internal)</pre>
<pre><font color="#4E9A06"><b>❯</b></font> sudo ./hidpp20-mouse-event-test /dev/hidraw1
Feature [0x8110] MouseButtonSpy unsupported.
Feature [0x8100] OnboardProfiles unsupported.
</pre>

<pre><font color="#CC0000"><b>❯</b></font> sudo ./hidpp-list-devices 
/dev/hidraw1: MX Master (046d:b012) HID++ 4.5
/dev/hidraw1 (device 0): MX Master (046d:b012) HID++ 4.5
</pre>
<pre><font color="#4E9A06"><b>❯</b></font> sudo ./hidpp-check-device /dev/hidraw1
4.5
</pre>

<pre><font color="#4E9A06"><b>❯</b></font> sudo ./hidpp20-onboard-profiles-get-description /dev/hidraw1 
terminate called after throwing an instance of &apos;HIDPP20::UnsupportedFeature&apos;
  what():  Feature [0x8100] OnboardProfiles unsupported.
Aborted
</pre>

> `event.function() is always DivertedButtonEvent, never DivertedRawXYEvent which I think should be the case for a gesture?`
>
> DivertedButtonEvent is called when a diverted button is pressed/released; [...]

PixlOne, 2019-11-13:
Hmm, I personally use the B01E/4071 variant of the MX Master, could it be that this bug is specific to the B012?

Does this bug also occur over USB? If so, could you use the [hidpp-dissector script on wireshark](https://github.com/libratbag/ratbag-toolbox/blob/master/dissectors/hidpp-dissector.lua) (from ratbag-toolbox) and check what (if any) HID++ events occurring while trying to use a gesture.

As far as I know, there is no way to use that dissector over bluetooth, so you will have to use the Unifying Receiver for this test.

kasra3422, 2020-06-20:
same issue here.
the gesture button doesn't work. "None" direction is working but other directions do nothing!

➜  tools git:(master) ✗ sudo ./hidpp-list-devices              
/dev/hidraw0: Logitech USB Receiver (046d:c52b) HID++ 1.0
/dev/hidraw0 (device 2): MX Master (046d:4041) HID++ 4.5

chamabreu, 2020-12-04:
> @PixlOne thanks for your working configuration. Is there any way to keep the "ALT+TAB" behaviour pressing (and keeping pressed) the 0xc3 button?

Hey did you get it? I am searching the answer for that question.

abraha2d, 2020-12-04:
@chamabreu Is https://github.com/PixlOne/logiops/issues/118 similar to what you're looking for?

chamabreu, 2020-12-10:
> @chamabreu Is #118 similar to what you're looking for?

kind of. I just throwed Linux over again, so it "solved" ;-) but if go in again, i will remember your answer and go deeper in testing. thanks buddy!