PixlOne/logiops #11: Installation problems

bodiroga, 2019-12-07:
Hi @PixlOne!

First of all, many thanks for developing this awesome tool, I'm looking forward to trying it with my new Logitech MX Master! :wink: 

I have, as far as I know, completed all the steps detailed in the "building" section of the readme, but I can see the following error in my systemctl status page:

```
dic 07 18:27:25 aitor-XPS-13-7390 systemd[1]: Started Logitech Configuration Daemon.
dic 07 18:27:25 aitor-XPS-13-7390 logid[32651]: /usr/local/bin/logid: error while loading shared libraries: libhidpp.so: cannot open shared object file: No such file or directory
dic 07 18:27:25 aitor-XPS-13-7390 systemd[1]: logid.service: Main process exited, code=exited, status=127/n/a
dic 07 18:27:25 aitor-XPS-13-7390 systemd[1]: logid.service: Failed with result 'exit-code'.
```

I'm executing the program on my Dell XPS 13 with Ubuntu 18.04 and the OS is new, so I had to install cmake, libevdev-dev, libudev-dev, libconfig-dev and libconfig++-dev (through 'apt install'), but the building and installation process went smoothly. The problem seems to be related with the installation of the 'hidpp' package, isn't it? How should I process?

Thank you for your help and keep up with the good job :+1: 

Best regards,

Aitor

PixlOne, 2019-12-10:
If you haven't installed libhidpp before, it should have installed libhidpp alongside it.

Could you find libhidpp.so in either /usr/local/lib/ or /usr/lib/? It's possible that you could work around this by symlinking /usr/lib/libhidpp.so to /usr/local/lib/libhidpp.so.

bodiroga, 2020-01-07:
Hi @PixlOne!

I don't know what I did (d'oh), but the libhidpp library was placed in the correct location without me doing anything consciously.

Thanks for your help!

closed 2020-01-07 by bodiroga

Rostislaved, 2020-04-21:
I found libhidpp.so in /usr/local/lib/ and copied it to /usr/lib/ and it helped. (ln -s didn't work)

`sudo cp /usr/local/lib/libhidpp.so /usr/lib/`

fourstepper, 2020-06-16:
Hi, after a succesful build on F32 I am getting the following errors, which I cannot solve by copying the libhidpp.so

~ ❯ sudo logid                  
logid: error while loading shared libraries: libhidpp.so: cannot open shared object file: No such file or directory

```
~ ❯ logid                       
logid: error while loading shared libraries: libhidpp.so: cannot open shared object file: No such file or directory

~ ❯ ls -la /usr/lib/libhidpp.so 
-rwxr-xr-x. 1 root root 2436376 Jun 16 19:22 /usr/lib/libhidpp.so

~ ❯ ls -la /usr/local/lib/libhidpp.so
-rwxr-xr-x. 1 root root 2436376 Jun 16 19:11 /usr/local/lib/libhidpp.so
```

Is there perhaps a different workaround that I could test? Thanks!

iamzubin, 2020-06-20:
Yeah, for some reason this issue happened recently and never happened before.
@PixlOne Did you change some linking config?
This is happening as the program is looking for files in lib64 instead of in lib.

```
[iamzubin@localhost projects]$ ldd /usr/bin/logid
    linux-vdso.so.1 (0x00007ffd053e9000)
    libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fd4d5282000)
    libevdev.so.2 => /lib64/libevdev.so.2 (0x00007fd4d5264000)
    libconfig++.so.11 => /lib64/libconfig++.so.11 (0x00007fd4d5247000)
    libhidpp.so => /lib64/libhidpp.so (0x00007fd4d50a8000)
    libudev.so.1 => /lib64/libudev.so.1 (0x00007fd4d507e000)
    libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007fd4d4e8e000)
    libm.so.6 => /lib64/libm.so.6 (0x00007fd4d4d46000)
    libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fd4d4d2b000)
    libc.so.6 => /lib64/libc.so.6 (0x00007fd4d4b61000)
    /lib64/ld-linux-x86-64.so.2 (0x00007fd4d52c1000)
```
You can simply make a linker to files from usr/local/lib to (I'm guessing local/lib64 should work), I simply copied all the files from lib to lib64

TL;DR
```
sudo mv /usr/lib/libhidpp.so /usr/lib64/
```

I'm not sure if this is the right way and might break later updates (If you update, I guess, Not a lot of people do)