PixlOne/logiops #13: Multiple devices

aleixq, 2020-01-08:
I have this scenario: One simple Logitech keyboard and mouse kit, and one mx master 3.
 If I have both connected logid outs:

```
# logid -v
[INFO] Found Logitech USB Receiver on /dev/hidraw2
terminate called without an active exception
```

So the configuration of mx master 3 is not applied.

If Logitech kit is **not** connected it outs:

 ```
# logid -v
[INFO] Missing hiresscroll option, not setting.
[WARN] No button configuration found, reverting to null config.
[INFO] MX Master 3 detected: device 255 on /dev/hidraw6
[INFO] MX Master 3 detected: device 0 on /dev/hidraw6
```

And is working as it should

PixlOne, 2020-01-12:
It seems that there was likely not joined/detached properly. It's likely somewhere in DeviceFinder.cpp, I'll try to debug this when I have the chance.

pacorayban, 2020-01-14:
In the same idea, when G-series receiver is plugged, I get a core dump.
If plugged before launching logid:
`# logid --verbose Debug
[INFO] Found Logitech USB Receiver on /dev/hidraw3
[INFO] Found Logitech Logitech BT Mini-Receiver on /dev/hidraw7
[INFO] Found Logitech USB Receiver on /dev/hidraw4
[DEBUG] Receiver on /dev/hidraw4: Device 1 paired
[INFO] Found Logitech USB Receiver on /dev/hidraw3
terminate called without an active exception
[1]    23198 abort (core dumped)  logid --verbose Debug`

If plugged after launching:
`# logid --verbose Debug
[INFO] Found Logitech Logitech BT Mini-Receiver on /dev/hidraw7
[INFO] Found Logitech USB Receiver on /dev/hidraw4
[DEBUG] Receiver on /dev/hidraw4: Device 1 paired
[INFO] Device MX Master 2S not configured, using default config.
[INFO] MX Master 2S detected: device 1 on /dev/hidraw4
[DEBUG] 0x800 was pressed but no action was found.
[ERROR] Time out on default device: /dev/hidraw6 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw6 (possibly asleep)
[INFO] Found Logitech USB Receiver on /dev/hidraw3
[INFO] Found Logitech USB Receiver on /dev/hidraw3
terminate called without an active exception
[1]    23042 abort (core dumped)  logid --verbose Debug
`

Malvineous, 2020-03-07:
I've found where this problem is, but I'm not quite sure how to fix it.  The problem is in this call:

https://github.com/PixlOne/logiops/blob/38dcc65d4b2692f2bd0bb5f39141ca57e54ce2ac/src/logid/DeviceFinder.cpp#L68-L73

Here, `index` already exists in the `std::map` which according to the docs means that the value being emplaced is constructed and then immediately destructed again.  Because this creates an `std::thread` then destroys it without calling `join()`, it calls `terminate()` and hence the crash.

I'm not sure why `insertNewReceiverDevice()` gets called twice for the same device - in my case there are calls for `hidraw3`, `hidraw4`, then `hidraw3` again.  However after the `hidraw4` call I get a message saying `Device 1 paired`, so it looks like after it finds a valid device it runs through all the hidraw devices again?

Malvineous, 2020-03-07:
Just to add to this, adding the following hack to `insertNewReceiverDevice()` fixes the problem, although of course this isn't the best solution, just confirmation of the problem:
```
Device* DeviceFinder::insertNewReceiverDevice(const std::string &path, HIDPP::DeviceIndex index)
{
    if (this->devices.find(path) != this->devices.end()) {
        log_printf(INFO, "%s already checked, ignoring", path);
        return nullptr;
    }

```
I guess you should be able to reproduce the problem at your end by temporarily tweaking the logic so that this function ends up getting called twice.

maxikd, 2020-10-23:
I recently got a MX Keys and I'm having trouble configuring both my MX Master 3 and MX Keys paired to the same Unifying receiver. Apparently, only one device is properly configured.
Any ideas what this might be?

Output:
`maximilian@desktop-max:~$ sudo logid
[INFO] Missing hiresscroll option, not setting.
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[INFO] MX Keys detected: device 2 on /dev/hidraw0
^C
maximilian@desktop-max:~$ sudo logid
[INFO] Missing hiresscroll option, not setting.
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[INFO] MX Master 3 detected: device 1 on /dev/hidraw0
^C
`

PixlOne, 2020-10-23:
Are you running the latest version of logid? Also, it doesn't seem like your keyboard is being picked up at all.

@maxikd

maxikd, 2020-10-23:
@PixlOne 
I just checked
```
maximilian@desktop-max:~/Documents/git/logid/build$ git pull
Already up to date.
maximilian@desktop-max:~/Documents/git/logid/build$ git log
commit 4c582241d5aed5a4657c119b6d85caaaf36df4ad (HEAD -> master, origin/master, origin/HEAD)
Merge: d1d8c49 132ae70
Author: pixl <pixlone@protonmail.com>
Date:   Thu Oct 8 23:33:49 2020 -0400

    Merge pull request #137 from gavinhungry/tested-mx-anywhere-3
    
    Add MX Anywhere 3 to TESTED.md

commit 132ae7025ac931131dac4fabecc65be951efe6ba
Author: Gavin Lloyd <gavinhungry@gmail.com>
Date:   Thu Oct 8 16:52:32 2020 -0700

    Add MX Anywhere 3 to TESTED.md

commit d1d8c49021466838a06b9109159d4aa7faffb17b
Merge: 951661d b728f36
Author: pixl <pixlone@protonmail.com>
Date:   Mon Sep 28 01:43:19 2020 -0400

    Merge pull request #136 from AvdN/patch-1
    
maximilian@desktop-max:~/Documents/git/logid/build$ cmake ..
LogiOps Version Number: v0.2.2-31-g4c58224
-- systemd units will be installed at /lib/systemd/system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/maximilian/Documents/git/logid/build
maximilian@desktop-max:~/Documents/git/logid/build$ make
[100%] Built target logid
maximilian@desktop-max:~/Documents/git/logid/build$ sudo make install
[100%] Built target logid
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/bin/logid
-- Up-to-date: /lib/systemd/system/logid.service
```
This means I'm on latest version, right?
When logid says it detected the MX Keys, the mappings I have configured work fine, but then the mouse's ones doesn't.

Do you have any suggestion of what to try?

PixlOne, 2020-10-23:
Is it possible that you have two versions of logid (one in `/usr/local/bin/logid` and another in `/usr/bin/logid`) and the old one is being referred to?

maxikd, 2020-10-24:
This may be possible. I'll clean all out and reinstall logid.
Just as info, if this happens, both the mouse and the keyboard work as intended on the config file:

```
maximilian@desktop-max:~$ sudo logid
[INFO] Missing hiresscroll option, not setting.
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[INFO] MX Master 3 detected: device 1 on /dev/hidraw0
[INFO] Connection established to MX Master 3
[INFO] MX Keys detected: device 2 on /dev/hidraw0
[INFO] Connection established to MX Master 3
```
This was after some hours of use.

maxikd, 2020-10-24:
@PixlOne that was it!! After deleting both logids and service, and reinstalling, it worked! And logid.service now works as it should. Before this I always got an exception.
Thank you very much! :D