PixlOne/logiops #14: OnFewPixels integer field pixels, needs some explanation

wiebel, 2020-01-16:
I tried to change the pixels field but failed to see any effect.
Also in Configuration.cpp line 412 is:
`log_printf(WARN, "Line %d: OnFewPixels requires a 'pixels' field.", gesture_config.getSourceLine());`
which suggests that using "OnFewPixels" demands the setting of pixels, which it doesn't.
It's running fine without pixxels set, but I'd like to tweak just that.

PixlOne, 2020-01-17:
Are you sure the gesture works? A gesture with the OnFewPixels mode shouldn't load if the pixels field is not set.

As it is right now, the configuration issues generally just print warnings and disable the individual component affected by the issue. They don't prevent the whole program from working.

wiebel, 2020-01-17:
As I have no idea in what context I have to set pixels and the whole config does bit contain the word pixels except for the OnFewPixels, it is not set, at least not 8n the the config. In any way it's working fine, I can switch my desktop multiple times during one gesture button press, it's just a bit to jumpy, if that makes any sense.

wiebel, 2020-01-20:
My logid.cfg is as  follows, and working fine incl. gestures:
```
{
    name: "MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 20;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 1500;
    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnFewPixels";
                        action =
                        {
                            type: "Keypress";
			    keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_UP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnFewPixels";
                        action =
                        {
                            type: "Keypress";
			    keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_DOWN"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnFewPixels";
                        action =
                        {
                            type: "Keypress";
			    keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnFewPixels";
                        action =
                        {
                            type: "Keypress";
			    keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_RIGHT"];
                        }
                    },
                    {
                        direction: "None"
                        mode: "NoPress"
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
}
);
```
Could you post an example hot the pixels field should be set?

wiebel, 2020-01-27:
Turns out the needed snipped was like:
```
...
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnFewPixels";
                        pixels: 10;
                        action =
...      
```

closed 2020-01-27 by wiebel