PixlOne/logiops #15: Direction dependen threshold for SmartShift

wiebel, 2020-01-16:
Now this is a long shot, but as I own the MX Revolution I might be bit spoiled here.
It was possible to have the threshold for the SmarShift threshold (name was different)  separately set for upwards and downward scrolling. This is incredibly comfortable as it's much easier to scroll down tha it is to scroll up. I have now idea if it's even possible with the hardware of the MX Master 3, but it would be awsome.

For reference see:
https://github.com/sebastien/revoco

PixlOne, 2020-01-17:
So currently, I'm using the native SmartShift feature on the mouse. As far as I know, the only way what you're suggesting could be done is by diverting scroll events to the program and listening for a large number of scroll events with different thresholds for up and down, and toggling SmartShift ratchet/freespin mode when the thresholds are reached and when the feel stops.

This seems a little tough to do, but I'll consider this as a feature suggestion. If anyone knows if there is a native way to set separate thresholds, please respond.

wiebel, 2020-01-17:
I'm pretty sure on the revolution it was a somewhat hidden feature if the hardware itself. Hidden, as it was not possible to set it up in the native windows app from logitech. I'm not so sure if.having it run inside the daemon is worth the trouble.

rockerbacon, 2020-01-23:
> I'm pretty sure on the revolution it was a somewhat hidden feature if the hardware itself. Hidden, as it was not possible to set it up in the native windows app from logitech. I'm not so sure if.having it run inside the daemon is worth the trouble.

Could possibly be added as device specific feature in the same fashion that gestures aren't available for all devices. A quick look a the linked repo does suggest that it's only using features available in the device itself.