PixlOne/logiops #16: Building logiops from zip file does not work

bolu-atx, 2020-01-20:
Hi, I'd like to report an issue with building logiops using the provided instructions on the Github readme.

If I download the ZIP file and try to build from source, the `git submodule checkout` for the dependency `libhidpp` fails and would not continue. I tried to build the master branch of `libhidpp` and installed it. It seems the include / lib directory structure have changed and logiops would get stuck complaining about the wrong include directory from `find_package(libhidpp)`.

Checking out master via git and then building does not have this issue.

Thanks!

rockerbacon, 2020-01-23:
When you download as a ZIP the compressed archive will not contain a `.git` folder, where git stores a bunch of information it needs. Without these informations git should not be expected to work properly, so it's no surprise that submodule management fails. This folder and its contents are automatically created when the repository is cloned.

I see two things which can be done:

1. Add information to the README that the project must be cloned: quick and easy
2. Provide prebuilt binaries (with libhidpp linked statically) so that cloning the project is only necessary for developers: relatively easy and will be necessary at some point. Requires changes to _CMakeLists.txt_

PixlOne, 2020-07-13:
As the refactor branch no longer has any submodules, I can safely say that this issue will be resolved in an upcoming update.

closed 2020-07-13 by PixlOne