PixlOne/logiops #18: MxMaster 2s or 3 support?

GitJamz, 2020-01-26:
Did anyone try logiops with mxmaster 2 or 3 ?

wiebel, 2020-01-27:
Sure, works perfectly wirh my MX Master 3.

GXGOW, 2020-01-30:
I'm also using logiops with the MX Master 3. Works like a charm!

adrfantini, 2020-02-06:
Can you map the thumb button with this tool, or is it still hardcoded with Ctrl+Alt+Tab?

GXGOW, 2020-02-06:
Sure, you can map cid's 0x53 (back button) and 0x56 (forward button) in your `logid.cfg` like this:

```
        {
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
```

adrfantini, 2020-02-06:
> Sure, you can map cid's 0x53 (back button) and 0x56 (forward button) in your `logid.cfg` like this:
>
> ```
>         {
>             cid: 0x53;
>             action =
>             {
>                 type : [...]

Thanks, but I meant the lower button where the thumb rests (which by default does Ctrl+Alt+Tab), not the two small buttons that are forward/back by default.

GXGOW, 2020-02-06:
> Thanks, but I meant the lower button where the thumb rests (which by default does Ctrl+Alt+Tab), not the two small buttons that are forward/back by default.
> 
> (edit: MXM3)

Oh, my mistake. I think you can set it the same way with cid `0xc3`. All other control IDs [are listed on the wiki](https://github.com/PixlOne/logiops/wiki/CIDs) by the way.

adrfantini, 2020-02-06:
That's awesome, it works flawlessly. Thanks!

Fourstepper, 2020-02-25:
Sorry to hijack, is it possible to make the thumb button + mouse movement change workspaces in GNOME/KDE, or still not possible?

Thank you!

GXGOW, 2020-02-26:
> Sorry to hijack, is it possible to make the thumb button + mouse movement change workspaces in GNOME/KDE, or still not possible?
> 
> Thank you!

That is possible, I have it configured for the up/down gestures like this for GNOME:

```
    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_PAGEUP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_PAGEDOWN"];
                        };
                    }
                );
            };
        },
   );

jef, 2020-05-20:
> I'm also using logiops with the MX Master 3. Works like a charm!

vthuongt, 2020-06-02:
The MX Master 2S works for me with Ubuntu 18.04 and following config:

```
devices: (
{
    name: "MX Master 2S";
    smartshift:
    {
        on: false;
        threshold: 15; # 7 is ideal for work
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 800;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT",  "KEY_UP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT"];
                        }
                    },

                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA"];
                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        }
    );
}
);
```
I can use gestures to switch workspaces in gnome and a single tap on the thumb button shows the overview for open windows (like pressing Win/Alt key).

jef, 2020-06-13:
I will say that it works just fine, but the main problem is that if and when it loses connection, I have to restart the `logid.service`. Not sure if it's just me or a side effect as of now.

tomekand11, 2021-01-08:
@jef does it happen on bluetooth only ?

jef, 2021-01-08:
> @jef does it happen on bluetooth only ?

I was using the dongle. That being said, I built from latest master a month or so after and didn't have any problems. I no longer have the mouse, but it seemed to be working fine then!

cerealfordinner, 2021-03-16:
Does anyone know how I would bind the thumb button to control + v (paste) to mimic how I use it Windows? Thank you in advance.

jef, 2021-03-16:
> Does anyone know how I would bind the thumb button to control + v (paste) to mimic how I use it Windows? Thank you in advance.

Off topic, but:

https://github.com/PixlOne/logiops/wiki/CIDs

Control ID | Function
-- | --
0x0052 | Middle Mouse Button

```
buttons: (
        {
            cid: 0x0052;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_V"];
            };
        }
    );
```

cerealfordinner, 2021-03-16:
> Off topic, but:
>
> https://github.com/PixlOne/logiops/wiki/CIDs
>
> Control ID	Function
> 0x0052	Middle Mouse Button
> ```
> buttons: (
>         {
>             cid: 0x0052;
>             action = [...]

Would this work for the thumb button on the side of the MX Master 2S?

whenever I run ```xev``` and capture the input it doesn't show as a button press which is what's messing me up with the binding.

```
KeyRelease event, serial 38, synthetic NO, window 0x3a00001,
    root 0x1de, subw 0x3a00002, time 25323386, (55,22), root:(2072,179),
    state 0x1c, keycode 64 (keysym 0xffe9, Alt_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 38, synthetic NO, window 0x3a00001,
    root 0x1de, subw 0x3a00002, time 25323396, (55,22), root:(2072,179),
    state 0x14, keycode 37 (keysym 0xffe3, Control_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

jef, 2021-03-26:
Your middle mouse button is probably a different CID. Could be `0x1de`. Not really sure how to read that output.

But once you find your correct CID, the rest should be fine.

adrian-huminiuc, 2023-01-09:
I wanted to bind the top button, (button behind the wheel ) to act as the middle button.
If it helps anyone here is the config, works with Wireless Mouse MX Master 3, ubuntu 22
```echo 'devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 10;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
       // mouse wheel click (old middle button) will toggle the smoothness of the scrolling
        {
            cid: 0x52;
            action =
            {
                type: "ToggleSmartShift";
            };
        },
        //top button (which toggled the smoothness of the scrolling) is now the middle button (like: closing tabs, copy pasta, etc)
        {
            cid: 0xc4;
            action =
            {
                type: "Keypress";
                keys: ["BTN_MIDDLE"];
            };
        }
    );
}
);' | sudo tee /etc/logid.cfg

PixlOne, 2023-05-03:
Closing this issue as these mice are supported.

closed 2023-05-03 by PixlOne