PixlOne/logiops #20: Terminated called after throwing an instance of 'HIDPP20::Error'

kevinsu628, 2020-01-29:
Hello, 

I am running into an error when I run "logid -v"

terminate called after throwing an instance of 'HIDPP20::Error'
  what():  Invalid function ID
Aborted

My Master MX 2S and keyboard USB receiver can both be successfully detected. 

Could you help me with the issue, please? 

Thank you in advance!

PixlOne, 2020-01-30:
Do you know what causes this error to appear?

asmith-r7, 2020-03-10:
I am also getting the same thing. I have my keyboard running on the unified receiver and my mouse over bluetooth

Jairo0120, 2020-03-11:
Maybe you have more than one logitech connected device. Try disconecting one, the keyboard could be the problem

BenjaminDuchadeuil, 2020-06-13:
I am having the same issue, I want to use logiops for my MX Master 2S, but I also have a Logitech Pro headset, and it's being recognized by the software, making it crash :
```
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[DEBUG] Receiver on /dev/hidraw0: Device 1 paired
[INFO] Device Logitech PRO not configured, using default config.
[INFO] Logitech PRO detected: device 255 on /dev/hidraw3
terminate called after throwing an instance of 'HIDPP20::Error'
  what():  Invalid feature index
Aborted
```

Of course, if I unplug my headset it will work, but as soon as I plug it back, the program crashes.
I've not looked into the code for now, but is there a way to ignore certain devices ?

PixlOne, 2020-06-15:
That's really odd @BenjaminDuchadeuil...

Can you compile [PixlOne/hidpp](https://github.com/PixlOne/hidpp), run the `hidpp-list-features` tool on the Logitech PRO headset (e.g. from the logs `hidpp-list-features -d 255 /dev/hidraw3` (`/dev/hidraw3` may differ).

BenjaminDuchadeuil, 2020-06-15:
The `features -d 255 /dev/hidraw3` command failed and returned `Invalid device index: 255` .
I tried without the index and got this :

```
Logitech PRO (046d:0aa7) is a HID++ 4.2 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x8310] Equalizer
Feature 0x05: [0x8300] Sidetone
```

PixlOne, 2020-07-04:
@BenjaminDuchadeuil I now have a Logitech PRO headset and I was testing out this bug and yeah, I can confirm that it's the same on my end.
```
terminate called after throwing an instance of 'HIDPP20::Error'
  what():  Invalid feature index
```
What's weird is that this is occurring after the program tries to determine whether it supports `[0x1b0x] Reprog controls`, on any other device, it just throw an `HIDPP20::UnsupportedFeature` error. I don't exactly know why this device behaves so strangely though.

This seems like a complex issue to solve and I'm already working on refactoring many parts of the program in the drop-libhidpp branch. So for now, I'll try to make a patch that'll allow you to disable devices if they match a certain PID.

PixlOne, 2020-07-04:
@BenjaminDuchadeuil Try the latest commit and add `blacklist: [0x00a7];` to your logid.cfg

BenjaminDuchadeuil, 2020-07-04:
This works perfectly. Thank you for your amazing work !

PixlOne, 2020-07-14:
This bug should be fixed in the refactored version. Also, @BenjaminDuchadeuil I can confirm that the G PRO headset works properly in the upcoming update. I'll re-implement the ignore feature for those who want it though.

closed 2020-07-15 by PixlOne