PixlOne/logiops #22: MX Anywhere 2S support?

elwin1234, 2020-02-01:
Hi,

I'm trying to find out if my (bluetooth) mouse, the **MX Anywhere 2S**, is (partly) working with **logiops**. It recognizes the mouse. Sees some unknown 'ghost' button pressed when starting the software? But does not seem to recognize any further button presses or gestures. Below the console output:

```bash
[23:16:02 e@dell ~$] sudo logid -v
[INFO] MX Anywhere 2S detected: device 255 on /dev/hidraw1
[ERROR] Device does not support SmartShift
[DEBUG] 0x800 was pressed but no action was found.
[INFO] MX Anywhere 2S detected: device 0 on /dev/hidraw1
[ERROR] Device does not support SmartShift
```

Furthermore it seems to start twice or something?

If I want to try to see if I can make it work where should I start?

Malvineous, 2020-03-08:
This looks fine, you just need to create a config file and it will work.

elwin1234, 2020-04-25:
Ah, yes, couldn't find the right CID's at first, they are not automatically printed in debug mode (-v) on mouse button presses. 

But found them in the wiki @ https://github.com/PixlOne/logiops/wiki/CIDs. 

Seems to work great now! Can confirm that **MX Anywhere 2S** is supported.

closed 2020-04-25 by elwin1234