PixlOne/logiops #23: Is Logitech Flow support planned?

Qwerty-Space, 2020-02-08:


PixlOne, 2020-02-09:
I personally can't work on that since I don't own any devices that support Logitech Flow.

Qwerty-Space, 2020-02-10:
Ah dang
I have an Mx Master 2s, which supports flow.  Is there any way I can help?

PixlOne, 2020-02-11:
Sure, if you use some of the tools on the hidpp repository and [this wireshark script](https://github.com/libratbag/ratbag-toolbox/blob/master/dissectors/hidpp-dissector.lua), you may be able to reverse engineer the Logitech Flow feature and somehow document it or write a feature addition.

Although I can't personally test any kind of support for Logitech Flow, I'd be open to including the feature if I can get others to test it out.

Qwerty-Space, 2020-02-16:
Reverse engineering sounds a bit advanced for me.

Malvineous, 2020-03-08:
@Qwerty-Space Is there a reason you would prefer Flow instead of using [Barrier](https://github.com/debauchee/barrier)? (an open source fork of the former Synergy project)

Qwerty-Space, 2020-03-11:
Oh I've not heard of that, will try

LiverBlow, 2020-09-21:
Cannot use barrier sadly due to security policy on my windows worklap, so a Linux version would be awesome still

Malvineous, 2020-09-21:
@LiverBlow A lot of work just to clone a proprietary version of Barrier/Synergy.  Might be easier to convince your admins to allow Barrier, if they have already allowed Logitech's equivalent!

barolo, 2020-09-21:
It's a 'change-host' functionality isn't it? Solaar has docs for it and it's implemented already, you can switch any amout of devices via command line instantly, Linux only obviously but it works very well. I have it scripted so my devices switch when touching left or right edge of the screen

Flobii, 2020-09-24:
I tried and am currently using Barrier, but I could not find a Flow like option to copy paste from Windows to Linux?

Malvineous, 2020-09-24:
Not sure, I just press Ctrl+C on Windows and Ctrl+V on Linux to copy and paste between machines.

Flobii, 2020-09-24:
oh, text works, thanks!

but files don't.

also: I'd be open to help/test Flow (I am too less of a genius for reverse engineering tho sadly)

PixlOne, 2020-10-02:
I still need to figure out how Logitech Flow works exactly. I know that the MX Master 2S and MX Master 3 support the Change Host feautre, and looking at past feature dumps, there doesn't seem to be anything firmware-specific about Logitech Flow.

I suspect the clipboard feature is really just some software-level trick like with Synergy. I also don't know if they're using the Change Host feature, since the delay with connection would be noticeable when moving between computers. The delay is noticeable on my MX Master at least, I don't know if the MX Master 2S and 3 have the same "issue".

Then again, this is low priority for me, since there are more established projects that bring similar functionality, and there are other features/bugs I need to work on here.

barolo, 2020-10-02:
At least with Solaar there's no delay, I have Craft [ MX keys ] and MX Master 2S, and they both switch instanly at once

barolo, 2020-10-19:
I would like to remind everyone that kdeconnect and its gnome offshots support clipboard sharing [ it has windows client even afaik ] Which allows clipboard syncing with almost any device transparently.

ashtez, 2021-01-23:
I don't know how barrier works, but Flow seems to disconnect your keyboard/mouse combo from one computer telling it to connect to another one.

Cherkah, 2021-02-22:
thus if someone is expecting install logiops to get benefits of the ' flow' feature hi is wrong?!!!

does someone can confirm, cause there is more than 1 week i'm digging the web to  have my mxMaster3 & mxKey works on my setup (2 archlinux) with flow

Ashark, 2021-11-14:
> @Qwerty-Space Is there a reason you would prefer Flow instead of using [Barrier](https://github.com/debauchee/barrier)? (an open source fork of the former Synergy project)

There are actually some good reasons to try/prefer Flow mode:
- Barrier currently does not support wayland (see https://github.com/debauchee/barrier/issues/1251)
- You want to avoid some issues of Barrier/Synergy. Currently there are many issues, such as: wrong handling of switching layouts (https://github.com/debauchee/barrier/issues/45), repeating keyup-keydown event instead of single keydown on button holding (https://github.com/symless/synergy-core/issues/6269) and so on.

jandestiny, 2021-12-29:
> It's a 'change-host' functionality isn't it? Solaar has docs for it and it's implemented already, you can switch any amout of devices via command line instantly, Linux only obviously but it works very well. I have it scripted so my devices switch when touching left or right edge of the screen

Hi, could you possibly provide me with that script for switching when touching the edges?

Ashark, 2021-12-29:
I also suffer right now because of lacking of this. I already asked @barolo, here is his answer: https://github.com/pwr-Solaar/Solaar/pull/844#issuecomment-968538385

In other words, there was no a touch edges solution in wayland. As a workaround, I used [lead](https://aur.archlinux.org/packages/lead-git/) app (on x11), and binded the command to switch.
Also, I am planning to make a proper kde screen edges solution (kwin script).
Also, I had a problem (did not explored yet) that if keyboard is sleeping, it is not switched with mouse. And then you got mouse on second host while keyboard on first host.

jandestiny, 2021-12-29:
> I also suffer right now because of lacking of this. I already asked @barolo, here is his answer: [pwr-Solaar/Solaar#844 (comment)](https://github.com/pwr-Solaar/Solaar/pull/844#issuecomment-968538385)
> 
> In other words, there was no a touch edges solution in wayland. As a workaround, I used [lead](https://aur.archlinux.org/packages/lead-git/) app (on x11), and binded the command to switch. Also, I am planning to make a proper kde screen edges solution (kwin script). Also, I had a problem (did not explored yet) that if keyboard is sleeping, it is not switched with mouse. And then you got mouse on second host while keyboard on first host.

Okay, that is frustrating :/ thank you for the fast reply though, it's much appreciated. Good luck with the kwin script. You should definitely reference it in this issue if you see any success with it!

barolo, 2021-12-29:
I don't have the corner script any longer as I prefer to switch via icon in the dock. I remember using [this](https://gitlab.com/divinae/toogleparachute/-/blob/master/contents/code/main.js) corner script as a base.

donrudo, 2023-08-10:
I am using [edges](https://github.com/MichaelStraube/edges) for the hot corners scripting, the issue is right now to make the mac or windows pc detect back the system as part of the Flow.
The script to trigger the change of host uses Solaar:
``` 
#!/usr/bin/env bash

 solaar config 'Mouse' change-host $1 &
 solaar config 'MX MCHNCL' change-host $1 &
```

jadaml, 2024-05-01:
> Sure, if you use some of the tools on the hidpp repository and [this wireshark script](https://github.com/libratbag/ratbag-toolbox/blob/master/dissectors/hidpp-dissector.lua), you may be able to [...]

**I volunteer, and offer all my help I can give.**

I have now a triage of Windows, Mac and Linux system. I've recently started using Kubuntu (with the intent on daily driving it) with Solaar. I only use Solaar to show the charge level though. I have an MX Keys and an MX Masters 3 though, I hope that will not be an issue.

In what setup do I need to run this Wireshark script, to be useful? I'm assuming you'd need me to showcase the original software's communication initially: For this I would run the script on one of Windows or Mac, and switch between the two device, then would try to copy some text then some file. What do I need to look for? What would be my task?

_I'm sure it is clear from the word_ volunteer, _that I may not be always available, but I_ (used to) _often disappear for a very long period of time._ Just to give you a fair warning. (Although I hope using Linux daily will break this habit of mine.)