PixlOne/logiops #24: Undefined symbol when running logid -v

Guara92, 2020-02-10:
Hi,

I installed logiops and all the dependencies, on ubuntu 19.10, including libhidpp, making a symlink /usr/local/lib/libhidpp.so to /usr/lib/libhidpp.so.
When i run logid -v this error is returned:
logid: symbol lookup error: logid: undefined symbol: _ZN7HIDPP109IReceiver6EventsE

What I'm missing?

Guara92, 2020-02-10:
I see now that libhidpp is a submodule of this project, I have not cloned the submodule and when logid did not start due to the lack of libhidpp, I installed it from its repository. Could this possibly create the problem?

PixlOne, 2020-02-10:
Did you use my fork of libhidpp? The original libhidpp repository is missing some features (e.g. the IReceiver feature which you're missing the symbol for).

Guara92, 2020-02-10:
no i used the original repo. I see in the CMakeLists.txt that the submodule should be fetched if missing but this has not happened.
If i try to init the submodule i have an error: 
>git submodule update --init -- hidpp
error: pathspec 'hidpp' did not match any file(s) known to git

PixlOne, 2020-02-11:
I'm not too sure why you're having that issue, but you manually clone this repo: https://github.com/PixlOne/hidpp

Guara92, 2020-02-11:
Thanks for the support, with your repo is working fine.

closed 2020-02-11 by Guara92