PixlOne/logiops #25: Mx Master 3 - side scroll wheel

dejdyla, 2020-02-25:
Hi,

I would like to map actions to side scroll wheel (located at left side of mx master 2 or 3). However I haven't found any **cid**, that represents this scroll wheel. 

Is there any possibility to map actions to side scroll wheel on mx master 3? 

Thanks in advance!

Malvineous, 2020-03-07:
I've done a little investigating and it looks like the mouse doesn't report scrolling events as button events, so there is no CID that you can use.  I am not sure whether it is technically possible to remap the scroll wheels, however Xbindkeys can bind actions to X11 mouse buttons and the side scroll wheel reports its motion as X11 buttons 6 and 7, so you might be able to use that in the mean time.

Fourstepper, 2020-03-08:
@dejdyla What's the use-case in mind? I would love to have this simply just for changing tabs in Chrome, but I am not sure how that is handled via Logitech Options on Windows/OSX

dejdyla, 2020-03-08:
> What's the use-case in mind? I would love to have this simply just for changing tabs in Chrome

That's also my use-case. 

> I am not sure how that is handled via Logitech Options on Windows/OSX

There are mappings per app (e.g. Chrome, Photoshop), which is more complex. I think that only 1 mapping for whole system (in our case:  "TAB+CTRL" to scroll right and "TAB+SHIFT+CTRL" to scroll left) would be sufficient.

leo-ventura, 2020-03-18:
@dejdyla @Fourstepper it is possible to achieve this by using xbindkeys and xte.

As better explained [here](https://wiki.archlinux.org/index.php/Logitech_MX_Master#Xbindkeys), you can map CTRL + TAB to scroll up and CTRL + SHIFT + TAB to scroll down.

All you have to do is create a `~/.xbindkeysrc` such as:
```bash
# thumb wheel up => change tab using ctrl tab
"xte 'keydown Control_L' 'keydown Tab' 'keyup Tab' 'keyup Control_L'"
   b:7 
                                        
# thumb wheel down => change tab backwards using ctrl shift tab
"xte 'keydown Control_L' 'keydown Shift_L' 'keydown Tab' 'keyup Tab' 'keyup Shift_L' 'keyup Control_L'"
   b:6
```

and then run `xbindkeys`. It's also probably a good idea to put xbindkeys as a startup application.

dejdyla, 2020-03-18:
@leo-ventura That is possible, but xbindkeys doesn't work under Wayland, does it?

Fourstepper, 2020-03-18:
@leo-ventura Thank you. Is it perhaps possible to map buttons and the scroll thumb wheel per application as well?

leo-ventura, 2020-03-18:
@dejdyla unfortunately, it doesn't. If you're using Sway, it may be possible to use [their own keybinding technique](https://wiki.archlinux.org/index.php/Sway#Custom_keybindings).

leo-ventura, 2020-03-18:
@Fourstepper apparently you can use [autokey](https://github.com/autokey/autokey) to map per application, but I haven't used it yet.