PixlOne/logiops #26: Cannot start logiops: libhidpp.so: cannot open shared object file: No such file or directory

asmith-r7, 2020-03-05:
`└[~]> systemctl status logid
● logid.service - Logitech Configuration Daemon
   Loaded: loaded (/lib/systemd/system/logid.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-03-05 10:28:51 GMT; 55min ago
  Process: 3013 ExecStart=/usr/local/bin/logid (code=exited, status=127)
 Main PID: 3013 (code=exited, status=127)

Mar 05 10:28:51 BFS-P500-3651 systemd[1]: Started Logitech Configuration Daemon.
Mar 05 10:28:51 BFS-P500-3651 logid[3013]: /usr/local/bin/logid: error while loading shared libraries: libhidpp.so: cannot open shared object file: No such file or directory
Mar 05 10:28:51 BFS-P500-3651 systemd[1]: logid.service: Main process exited, code=exited, status=127/n/a
Mar 05 10:28:51 BFS-P500-3651 systemd[1]: logid.service: Failed with result 'exit-code'.
`

`ls /usr/local/lib     
libhidpp.so  node_modules  python2.7  python3.7`

Is this the correct location for the libhidpp library?

PixlOne, 2020-03-06:
try symlinking /usr/lib/libhidpp.so to /usr/local/lib/libhidpp.so

asmith-r7, 2020-03-09:
The symlinking worked, but now it seems the driver is crashing for another reason

```
└[~]> sudo logid -v
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[DEBUG] Receiver on /dev/hidraw0: Device 1 paired
[INFO] Device K750 not configured, using default config.
[INFO] K750 detected: device 1 on /dev/hidraw0
[INFO] MX Master detected: device 255 on /dev/hidraw2
terminate called after throwing an instance of 'HIDPP20::Error'
  what():  Invalid function ID
[1]    5707 abort      sudo logid -v
```

GitJamz, 2020-03-19:
I got the same error:
logid[14239]: /usr/local/bin/logid: error while loading shared libraries: libhidpp.so: cannot open shared object file: No such file or directory

Something is wrong here..

Elijas, 2020-04-24:
Got the same errors on Ubuntu 18.04.

`libconfig.h ++ does not exist`
got fixed with
`$ sudo apt install cmake libconfig ++-dev`
and 
the `libhidpp.so: cannot open shared object file` 
got solved with
`$ sudo ln -s /usr/local/lib/libhidpp.so /usr/lib/libhidpp.so`
[source](https://www.st1t.com/entry/2020/03/10/090000)

p-himik, 2020-05-11:
A better workaround is to replace
```
ExecStart=/usr/local/bin/logid
```
with
```
ExecStart=env LD_LIBRARY_PATH=/usr/local/lib /usr/local/bin/logid
```
to avoid changing the state of the rest of your system.

@PixlOne Since it's about the `logid.service` file, I want to also mention that `StartLimitIntervalSec` is in the wrong section. It has to be specified in `[Unit]`, not in `[Service]`. Apart from that, great software, thank you!

PixlOne, 2020-05-15:
Yeah, I fixed this by using /usr/ as CMAKE_INSTALL_PREFIX instead of /usr/local.

Let me know if this solved it.

commit 8de0f7e0 (2020-05-15): Set CMAKE_INSTALL_PREFIX to /usr

Should fix #26 and (hopefully) any questions relating to missing
libhiddp.so.
closed 2020-05-15 by PixlOne