PixlOne/logiops #28: Consider linking to example configuration file on the 'Configuration' page in the wiki

ChrisAJS, 2020-03-12:
Whilst trying to figure out how to configure logiops, I found the [configuration wiki page](https://github.com/PixlOne/logiops/wiki/Configuration) which goes into detail as to how a configuration file should be laid out, but it doesn't have any examples.

I think linking to the [example configuration file](https://github.com/PixlOne/logiops/blob/master/logid.example.cfg) in the repo right at the top of the page would be a great idea- especially if you use this file for testing logiops during development, as if the wiki becomes out of date (as the disclaimer warns) at least the example should still be up to date.

P.S. Amazing project! 👍

PixlOne, 2020-03-18:
It seems that I forgot to do that, thank you.

closed 2020-03-18 by PixlOne

Fourstepper, 2020-03-19:
@ChrisAJS Also just wanted to say that there are a few examples and a little guide on the usage also on the Arch Wiki https://wiki.archlinux.org/index.php/Logitech_MX_Master