PixlOne/logiops #29: OnFewPixels seems to only work once

muesli, 2020-03-18:
I'm trying to hook up my thumb button to react to directional moves while pressed, like in the example configuration. That works fine, but when I try to switch from `OnRelease` mode to `OnFewPixels` the action seems to only ever be triggered once, until the next restart of `logid`.

PixlOne, 2020-03-18:
That's weird, my MX Master doesn't seem to have that issue.

Could you post a log of the program and your config file?

muesli, 2020-04-01:
Config:

```
devices: (
{
    name: "MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 10;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 1250;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnFewPixels";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_PAUSE"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_F12"];
                        };
                    },
                    {
                        direction: "None";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";   
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_TAB"];
                        };
                    }
                );
            };
        }
    );
}
);
```

`KEY_PAUSE` triggers my Terminal being focused on my system. When I hold the thumb button and move the mouse up the terminal quickly gains focus, then loses it again. It looks as if the event is triggered twice in very quick succession - and then never again. The other actions keep working.

PixlOne, 2020-04-01:
Your config file was missing a `pixels` field for OnFewPixels which resulted in undefined behaviour. It should have warned you about that and disabled the gesture but for some reason, it didn't.

I just fixed this in the latest commit.

closed 2020-04-01 by PixlOne

muesli, 2020-04-01:
@PixlOne Oh, I see. Thank you! I also misunderstood the function: I thought it'd only trigger once (after a "few" pixels). Maybe such a "single-shot" method could be added as well?

PixlOne, 2020-04-01:
Yeah, I didn't really know what I should call it. In hindsight, I should have called it "PixelInterval". I might change it but I'd like to avoid messing with people's configs.

muesli, 2020-04-01:
I agree, the config should probably remain compatible. We could either add a separate `singleshot` mode, or switch to that behavior in `OnFewPixels` mode when `pixels` is set to 0?

muesli, 2020-04-01:
Also, maybe it would be a nice idea to add a `OnFewPixels` mode configuration to the example configuration file?