PixlOne/logiops #30: CID of the DPI button of the MX Vertical

maxfrei750, 2020-03-18:
Hi, thanks for your awesome project!

I have an MX Vertical mouse and would like to bind the DPI button to Media Pause/Start. Unfortunately, I could not find the CID for the DPI button anywhere.

#3 mentions `hidpp-list-devices`. However, I could not find it in the repos of Manjaro or Arch. Do I have to build it from source or is it included somewhere with logiops?

Thanks!

PixlOne, 2020-03-18:
The hidpp tools are in [my hidpp repo](https://github.com/PixlOne/hidpp). I'm considering adding a cmake flag to build the hidpp tools in addition to libhidpp.

JCQuintas, 2020-03-21:
We could have a sort of debug mode where every key press logs it's CID.

maxfrei750, 2020-03-21:
@JCQuintas I think that would be a very good idea!

fstreun, 2020-05-14:
@maxfrei750, the CID of the DPI button of the MX Vertical is 0xFD (described in the wiki https://github.com/PixlOne/logiops/wiki/CIDs with "Mouse Sensitivity?").

`hidpp20-reprog-controls --device .... info` showed by following output:
```
Control Task    Info    Fn      Capabilities    Group   Remap to groups
0x0050  0x0038  mouse   -       reprog  1       1
0x0051  0x0039  mouse   -       reprog  1       1
0x0052  0x003a  mouse   -       reprog, divert, persist 2       1+2
0x0053  0x003c  mouse   -       reprog, divert, persist 2       1+2
0x0056  0x003e  mouse   -       reprog, divert, persist 2       1+2
0x00fd  0x00d2  mouse   -       reprog, divert, persist 2       1+2
0x00d7  0x00b4  -       -       divert, virtual 3       -
```

All the CIDs are correctly described in the wiki. Only the last line is a mystery to me.

PixlOne, 2020-05-15:
The MX Master has that button as well, iirc that CID refers to the button at the bottom of the mouse that allows you to switch between receivers.

I don't exactly know why I wrote 'Windows Navigation' in the wiki but I'll be sure to update it.

maxfrei750, 2020-05-15:
@fstreun Thanks!

closed 2020-05-15 by maxfrei750