PixlOne/logiops #31: Parse error on Line 11: syntax error

Fourstepper, 2020-03-19:
Hello, asking for help here, can't seem to figure out why there is an error on the mentioned line

```
devices: ({
  name: "MX Master 3";

  // A lower threshold number makes the wheel switch to free-spin mode
  // quicker when scrolling fast.
  smartshift: { on: true; threshold: 15; };
	buttons: (
	    // Make top button (0xc4) switch modes
	    { cid: 0xc4; action = { type = "ToggleSmartshift"; }; },

  hiresscroll: { hires: true; invert: false; target: true; };

  // Higher numbers make the mouse more sensitive (cursor moves faster),
  // 4000 max for MX Master 3.
  dpi: 1270;

  buttons: (

    // Make thumb button 10.
    { cid: 0xc3; action = { type: "Keypress"; keys: ["BTN_FORWARD"]; }; },

    // Make top button 11.
    { cid: 0xc4; action = { type: "Keypress"; keys: ["BTN_BACK"];    }; },

  );
});

```
>> This is the line in question:  hiresscroll: { hires: true; invert: false; target: true; };

Thank you for letting me know what coudl be wrong...

PixlOne, 2020-04-01:
```
buttons: (
	    // Make top button (0xc4) switch modes
	    { cid: 0xc4; action = { type = "ToggleSmartshift"; }; },
```
should be removed, you have another section for buttons below and this one is improperly closed.

Your config file should look like this:
```
devices: ({
  name: "MX Master 3";

  // A lower threshold number makes the wheel switch to free-spin mode
  // quicker when scrolling fast.
  smartshift: { on: true; threshold: 15; };

  hiresscroll: { hires: true; invert: false; target: true; };

  // Higher numbers make the mouse more sensitive (cursor moves faster),
  // 4000 max for MX Master 3.
  dpi: 1270;

  buttons: (

    // Make thumb button 10.
    { cid: 0xc3; action = { type: "Keypress"; keys: ["BTN_FORWARD"]; }; },

    // Make top button 11.
    { cid: 0xc4; action = { type: "Keypress"; keys: ["BTN_BACK"];    }; },

  );
});
```

PixlOne, 2020-04-01:
Also, regarding horizontal scrolling while holding Shift, do you mean that you'd like to change what the horizontal scroll wheel does when Shift is held?

Logid currently doesn't support modifying the horizontal scrollwheel's action but I'm thinking of adding it as a feature.

closed 2020-04-01 by PixlOne

Elijas, 2020-04-24:
> Logid currently doesn't support modifying the horizontal scrollwheel's action but I'm thinking of adding it as a feature.

This would be great

benjezzy, 2020-06-05:
+1 on the request of supporting horizontal scrollwheel

Elijas, 2020-06-06:
> +1 on the request of supporting horizontal scrollwheel

There is a workaround you can use for remapping horizontal scrollwheel.
Create file `~/.xbindkeysrc` with contents
```
"amixer -D pulse sset Master 3%-"
  b:6
"amixer -D pulse sset Master 3%+"
  b:7
```
`b:6` means horizontal-wheel-scroll-down, and `b:7` is scroll-up. You can use any command, but the `amixer` command here controls sound volume