PixlOne/logiops #32: [Ubuntu] Could not create evdev device: Permission denied

JCQuintas, 2020-03-20:
When I run `logid` it logs the following error: 

```[ERROR] Could not create evdev device: Permission denied```

It works when I use `sudo` though, so I wanted to know if there is any way for me to bypass having to use `sudo`

PixlOne, 2020-03-20:
Try adding the user to the `plugdev` group and logging back in.

`# usermod -a -G plugdev username`

JCQuintas, 2020-03-21:
Doesn't seem to have any effect.

MatthijsvE, 2020-04-20:
I have the same problem, the usermod command does not seem to have an effect at my system (Ubuntu 19.10) either. Is there something else that might work?

pascal-mack, 2020-05-29:
Similar Error, mine says:
[ERROR] Could not create evdev device: No such file or directory.
Sudo doesn't help

nylon86, 2020-07-03:
same problem of @pascal-mack on manjaro linux

maxikd, 2020-07-20:
Same on Pop!_OS 20.04. Works fine with `sudo`

StefanLobbenmeier, 2020-10-24:
From my understanding of linux so far, what you want to do is to check the permissions on the devices that are used, in my case:
```
WARN] Error adding device /dev/hidraw0: open failed: Permission denied
[WARN] Error adding device /dev/hidraw1: open failed: Permission denied
[WARN] Error adding device /dev/hidraw3: open failed: Permission denied
[WARN] Error adding device /dev/hidraw2: open failed: Permission denied
[WARN] Error adding device /dev/hidraw5: open failed: Permission denied
[WARN] Error adding device /dev/hidraw6: open failed: Permission denied
^C
➜  ~ ls -ashl /dev/hidraw*
0 crw------- 1 root root 237, 0 24. Okt 10:58 /dev/hidraw0
0 crw------- 1 root root 237, 1 24. Okt 10:58 /dev/hidraw1
0 crw------- 1 root root 237, 2 24. Okt 10:58 /dev/hidraw2
0 crw------- 1 root root 237, 3 24. Okt 10:58 /dev/hidraw3
0 crw------- 1 root root 237, 5 24. Okt 10:58 /dev/hidraw5
0 crw------- 1 root root 237, 6 24. Okt 10:58 /dev/hidraw6
```

You can see that the devices are owned by root and the group root. You will have to change something about this to allow non-root users to access those devices. I will check the wiki of ddcutil what the common way to change this is.

StefanLobbenmeier, 2020-10-24:
So using https://www.ddcutil.com/config/#grant-readwrite-permission-for-the-devi2c-n-devices-representing-monitors as a reference what you want to do is:

# How to resolve permission denied for non-root users

1. Add a file with the following line to /etc/udev/rules.d (in case of manjaro / arch you want to create a new file in /etc/udev/rules.d)
`sudo nano /etc/udev/rules.d/46-logiops-hidraw.rules`
2. content:
```
# Assigns the hidraw devices to group hidraw, and gives that group RW access:
KERNEL=="hidraw-[0-9]*", GROUP="hidraw", MODE="0660"
```
3. Add the group hidraw if it does not already exist:
`sudo groupadd --system i2c`
4. Add the users who will run logid to that group:
`usermod -G hidraw -a <yourname>`

StefanLobbenmeier, 2020-10-24:
Afterwards you should see that the devices have the correct group and you no longer get permission denied:
```
➜  ~ ls -ashl /dev/hidraw*
0 crw-rw---- 1 root hidraw 237, 0 24. Okt 11:32 /dev/hidraw0
0 crw-rw---- 1 root hidraw 237, 1 24. Okt 11:32 /dev/hidraw1
0 crw-rw---- 1 root hidraw 237, 2 24. Okt 11:32 /dev/hidraw2
0 crw-rw---- 1 root hidraw 237, 3 24. Okt 11:32 /dev/hidraw3
0 crw-rw---- 1 root hidraw 237, 5 24. Okt 11:32 /dev/hidraw5
0 crw-rw---- 1 root hidraw 237, 6 24. Okt 11:32 /dev/hidraw6
0 crw-rw---- 1 root hidraw 237, 7 24. Okt 11:32 /dev/hidraw7
0 crw-rw---- 1 root hidraw 237, 8 24. Okt 11:32 /dev/hidraw8
➜  ~ logid
[ERROR] I/O Error while reading /etc/logid.cfg: FileIOException
[INFO] Detected receiver at /dev/hidraw2
[INFO] Device Wireless Mouse MX Master not configured, using default config.
[INFO] Device found: Wireless Mouse MX Master on /dev/hidraw2:1
[INFO] Device Logitech Gaming Keyboard G610 not configured, using default config.
[INFO] Device found: Logitech Gaming Keyboard G610 on /dev/hidraw1:255
[WARN] Error adding device /dev/hidraw3: std::exception
^C
```
@JCQuintas please let us know if that fixed you issue
@PixlOne please consider adding this somewhere in the wiki / readme

yrd, 2021-09-26:
In case someone else stumbles on this issue with the same problem: the logid process also needs write access to `/dev/uinput`. I'm not sure if this is documented anywhere else, but I found it [here](https://hackage.haskell.org/package/evdev#permissions).

My complete udev config now looks like this:

```
KERNEL=="uinput", GROUP="input", MODE:="0660", OPTIONS+="static_node=uinput"
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="046d", MODE="0660", GROUP="input"
```

I'm using the `input` group instead of `hidraw`, as Stefan did. Also it might be worth noting that I don't use Ubuntu, so it may have worked for the previous commenters because Ubuntu could already be ships something like the first rule.