PixlOne/logiops #33: Can't manage to start daemon at boot.

PersonalDavid, 2020-03-20:
Hello! Thank you for your amazing work!!
For me, everything works just fine, its that i have to start up a terminal and type in 
`$ sudo logid`
in order to make it work, on every single start.
I've tried as advised:
`$ sudo systemctl start logid`
but unfortunately it just wont start on boot. 
I'm not a huge linux mage, and sadly can't figure out the problem by myself. :disappointed:

therealmarv, 2020-03-20:
```
sudo systemctl enable logid
```

should do it. It creates a symlink for systemd startup.

PersonalDavid, 2020-03-20:
Fantastic, the command did the magic. Thank you for your time and also a huge thanks for the driver again, you're saving the bottom of several 100$ rodent that doesn't work properly on linux!! 
Cheers

closed 2020-03-20 by PersonalDavid