PixlOne/logiops #34: connection issues, MX Master 2S not being found

crunk1, 2020-03-22:
Hey, great job in providing this driver. I'm getting an intermittent issue finding my device. My logid.cfg in my examples below simply maps NumLock to my gesture button.

I successfully ran this successfully earlier and it located my 2S. Though, it seemed like it would "rediscover" my 2S occasionally (which I think it would make my mouse glitch for a sec).

Then I killed logid. And now I'm 0/5 in getting logid to find my 2S (this is where I guess I might have to restart).

Successful run (with those rediscoveries):
```
> sudo logid -c /etc/logid.cfg
[INFO] Missing hiresscroll option, not setting.
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[INFO] MX Master 2S detected: device 3 on /dev/hidraw0
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[INFO] Connection established to MX Master 2S
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[INFO] Connection established to MX Master 2S
...
```

Example failing run (with -v, you can see my 2S -- Device 3)
```
> sudo logid -c /etc/logid.cfg -v
[INFO] Missing hiresscroll option, not setting.
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[DEBUG] Receiver on /dev/hidraw0: Device 3 paired
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
[ERROR] Time out on default device: /dev/hidraw7 (possibly asleep)
...
```

PixlOne, 2020-03-22:
When the mouse is not used for a while, it goes to sleep and times out. Usually, this can be fixed by just shaking the mouse or clicking to wake it.

Are these "errors" occurring when the mouse is in use?

crunk1, 2020-03-24:
Yeah, I'm using the mouse while it is running. Inactivity might explain the reconnections in the first output, but in output #2, I'm using the mouse the whole time.

vthuongt, 2020-03-24:
I also had this issue of a glitching mouse during usage, which however was unrelated to the driver. The mouse was connected via the Unify-dongle and it had connections problems leading to occasional reconnections when the dongle was placed near another used USB3 port. I stumbled over [this post](https://www.reddit.com/r/buildapc/comments/7ouumm/usb_30_with_logitech_unify_receiver_causing/) which explained that there was an issue with leaking usb 3 frequencies interfering with the unify-dongle. Solution was simply to move the dongle further away from those leaking usb ports. Hope that help.

crunk1, 2020-03-25:
The receiver is pretty close to the mouse (plugged into the back of my wired keyboard). The first example isn't that big of an issue anyway. The second example is more annoying -- when I can't get logid to find the device at all (even though it detects that it is paired).