PixlOne/logiops #35: Few pixels doesn't prevent OnRelease from triggering

T99Rots, 2020-03-25:
I currently have my config set to few pixels for up and down gestures controlling volume and to OnRelease for left and right controlling previous and next song, this does work but has the downside that while moving even a little bit to much to the left or right while adjusting the volume it will skip to the previous or next song, my suggestion would be to detect the axis on which a gesture is moving as soon as possible (after having moved n amount of pixels, maybe configurable) and then after this detection prevent gestures defined on a different axis from triggering.

My configuration file for reference:

```
devices: (
{
    name: "MX Master 3";
    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnFewPixels";
                        pixels: 50;
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_VOLUMEUP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnFewPixels";
                        pixels: 50;
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_VOLUMEDOWN"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_PREVIOUSSONG"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_NEXTSONG"]
                        }
                    },
                    {
                        direction: "None";
                        mode: "OnRelease";
                        action = 
                        {
                            type: "Keypress";
                            keys: ["KEY_PLAYPAUSE"];
                        }
                    }
                );
            };
        }
    );
}
);
```

abraha2d, 2020-11-24:
^ I've addressed locking gestures to a single axis on that PR. Try it out, let me know if it works.

T99Rots, 2020-12-06:
> ^ I've addressed locking gestures to a single axis on that PR. Try it out, let me know if it works.

Hey, i just got around to trying it and it works really well!