PixlOne/logiops #36: Enable Logid On Boot

Mahooki, 2020-03-29:
Hello and thank you for your time working on this project, I appreciate it.

I'm having issues when enabling logid.service on boot. I am running on Arch Linux. It functions as expected when I manually start it after logging in. For example:
```bash
systemctl start logid
```

# Enabling the Service

## Expected

Running `systemctl enable logid` will cause logid to start at boot successfully.

## Result

```bash
$ sudo systemctl enable logid
Created symlink /etc/systemd/system/multi-user.target.wants/logid.service → /usr/lib/systemd/system/logid.service.
$ sudo reboot
```

After enabling the service and rebooting, `journalctl` shows a `systemd-coredump` along with the following:
```text
logid.service: Main process exited, code=killed, status=6/ABRT
logid.service: Failed with result 'signal'.
SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=logid comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
```

# Enabling the Service (--user)
I have also tried enabling logid as a user service with the following contents

~/.config/systemd/user/logid.service
```text
[Unit]
Description=Logitech Configuration Daemon

[Service]
Type=simple
ExecStart=/usr/bin/logid
ExecReload=/bin/kill -HUP $MAINPID

[Install]
WantedBy=default.targe
```

## Expected

Running `systemctl --user enable logid` will cause logid to start at user login successfully.

## Result

```bash
$ sudo systemctl --user enable logid
Created symlink /etc/systemd/system/multi-user.target.wants/logid.service → /usr/lib/systemd/system/logid.service.
$ sudo reboot
```

After enabling the user service and rebooting, `journalctl --user` shows the following:
```text
[ERROR] Could not create evdev device: Permission denied
logid.service: Main process exited, code=exited, status=1/FAILURE
logid.service: Failed with result 'exit-code'.
```

Forgive me if I am missing something. I am happy to help debug and try to solve the issue if others are also experiencing this. Thank you for your time.

Mahooki, 2020-03-29:
Closing for now as it appears to be related to a second Logitech USB mouse receiver (G603) which was also plugged in. After removing it, leaving only the Unifying receiver, the standard `systemctl enable logid` appears to be running as expected. I even added the G603 receiver back in and rebooted - still appears to be working.

PixlOne, 2020-03-30:
Hmm, that shouldn't be happening. Having two unifying receivers work for me. Is this just an issue with the G603?

Mahooki, 2020-03-31:
The strange part is that it was a consistent issue before unplugging the g603, but once I unplugged it once it seems to be resolved even when I have it plugged back in. I'm not sure what would cause that behavior though. Maybe it had to do with usb device naming? shot in the dark.

It seems to be resolved now, so with both receivers plugged in I am not having issues. Below is the output of running lsusb when things are working for what its' worth.

```text
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 08bb:2902 Texas Instruments PCM2902 Audio Codec
Bus 003 Device 003: ID b58e:9e84 Blue Microphones Yeti Stereo Microphone
Bus 003 Device 002: ID 04d9:0141 Holtek Semiconductor, Inc. USB Keyboard
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0451:8140 Texas Instruments, Inc. TUSB8041 4-Port Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 1b1c:0c15 Corsair H100i Platinum
Bus 001 Device 006: ID 1b1c:0c0b Corsair Lighting Node PRO
Bus 001 Device 004: ID 05ac:0267 Apple, Inc. Magic Keyboard A1644
Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 0451:8142 Texas Instruments, Inc. TUSB8041 4-Port Hub
Bus 001 Device 002: ID 046d:c539 Logitech, Inc. USB Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

PixlOne, 2020-03-31:
So having both receivers connected only break logid when it starts on boot?

Are you running any other programs that interact with the G603 on startup?

Mahooki, 2020-03-31:
It appears intermittent, I restarted a few times until I got an error with it starting. I don't believe I am running anything else that interacts with either mouse on startup (at least, nothing in particular). Here is the journalctl output for the last boot which had an issue:
<details>
<pre>
-- Logs begin at Wed 2020-03-25 21:14:56 EDT, end at Mon 2020-03-30 20:30:34 EDT. --
Mar 30 20:28:43 archlinux kernel: Linux version 5.4.28-1-lts (linux-lts@archlinux) (gcc version 9.3.0 (Arch Linux 9.3.0-1)) #1 SMP Wed, 25 Mar 2020 13:04:22 +0000
Mar 30 20:28:43 archlinux kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-linux-lts root=UUID=a326107c-df79-4b8b-ba76-ededadaeec2d rw loglevel=3 quiet
...
Mar 30 20:28:43 archlinux systemd[1]: Reached target System Initialization.
Mar 30 20:28:43 archlinux systemd[1]: Started Daily Cleanup of Temporary Directories.
Mar 30 20:28:43 archlinux systemd[1]: Reached target System Time Set.
Mar 30 20:28:43 archlinux systemd[1]: Reached target System Time Synchronized.
Mar 30 20:28:43 archlinux systemd[1]: Started Daily man-db regeneration.
Mar 30 20:28:43 archlinux systemd[1]: Started Daily verification of password and group files.
Mar 30 20:28:43 archlinux systemd[1]: Reached target Timers.
Mar 30 20:28:43 archlinux systemd[1]: Listening on D-Bus System Message Bus Socket.
Mar 30 20:28:43 archlinux systemd[1]: Reached target Sockets.
Mar 30 20:28:43 archlinux systemd[1]: Reached target Basic System.
Mar 30 20:28:43 archlinux systemd[1]: Started D-Bus System Message Bus.
Mar 30 20:28:43 archlinux audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=dbus comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Mar 30 20:28:43 archlinux systemd[1]: Starting Network Manager...
Mar 30 20:28:43 archlinux systemd[1]: Started Logitech Configuration Daemon.
<b>Mar 30 20:28:43 archlinux audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=logid comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Mar 30 20:28:43 archlinux kernel: audit: type=1130 audit(1585614523.626:9): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-timesyncd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Mar 30 20:28:43 archlinux kernel: audit: type=1130 audit(1585614523.636:10): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=dbus comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Mar 30 20:28:43 archlinux kernel: random: dbus-daemon: uninitialized urandom read (12 bytes read)
Mar 30 20:28:43 archlinux systemd[1]: Starting Home Area Manager...
Mar 30 20:28:43 archlinux systemd[1]: Starting Login Service...
Mar 30 20:28:43 archlinux kernel: random: dbus-daemon: uninitialized urandom read (12 bytes read)
Mar 30 20:28:43 archlinux NetworkManager[553]: <info>  [1585614523.6720] NetworkManager (version 1.22.10-1) is starting... (for the first time)
Mar 30 20:28:43 archlinux NetworkManager[553]: <info>  [1585614523.6720] Read config: /etc/NetworkManager/NetworkManager.conf (lib: 20-connectivity.conf)
Mar 30 20:28:43 archlinux systemd[1]: Started Network Manager.
Mar 30 20:28:43 archlinux audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Mar 30 20:28:43 archlinux NetworkManager[553]: <info>  [1585614523.6730] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
Mar 30 20:28:43 archlinux systemd[1]: Reached target Network.
Mar 30 20:28:43 archlinux systemd[1]: Starting Permit User Sessions...
Mar 30 20:28:43 archlinux NetworkManager[553]: <info>  [1585614523.6767] manager[0x55ab7eb36080]: monitoring kernel firmware directory '/lib/firmware'.
Mar 30 20:28:43 archlinux dbus-daemon[552]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.12' (uid=0 pid=553 comm="/usr/bin/NetworkManager --no-daemon ")
Mar 30 20:28:43 archlinux systemd[1]: Finished Permit User Sessions.
Mar 30 20:28:43 archlinux audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-user-sessions comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Mar 30 20:28:43 archlinux systemd[1]: Starting GNOME Display Manager...
Mar 30 20:28:43 archlinux systemd[1]: Starting Hostname Service...
Mar 30 20:28:43 archlinux systemd[1]: Started GNOME Display Manager.
Mar 30 20:28:43 archlinux audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=gdm comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Mar 30 20:28:43 archlinux dbus-daemon[552]: [system] Activating via systemd: service name='org.freedesktop.login1' unit='dbus-org.freedesktop.login1.service' requested by ':1.13' (uid=0 pid=562 comm="/usr/bin/gdm ")
Mar 30 20:28:43 archlinux systemd[1]: Condition check resulted in FUSE Control File System being skipped.
Mar 30 20:28:43 archlinux systemd[1]: Condition check resulted in Load Kernel Module drm being skipped.
Mar 30 20:28:43 archlinux systemd[1]: Condition check resulted in First Boot Wizard being skipped.
Mar 30 20:28:43 archlinux systemd[1]: Condition check resulted in File System Check on Root Device being skipped.
[... cut ...]
Mar 30 20:28:49 archlinux /usr/lib/gdm-x-session[961]: Errors from xkbcomp are not fatal to the X server
Mar 30 20:28:49 archlinux systemd[1]: logid.service: Main process exited, code=killed, status=6/ABRT
Mar 30 20:28:49 archlinux systemd[1]: logid.service: Failed with result 'signal'.
Mar 30 20:28:49 archlinux audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=logid comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'</b>
...
</pre>
</details>

kashing1999, 2020-04-01:
I happen to have the same problem but unlike @Mahooki I only have my MX Master 3 plugged in.

[Journalctl output of where the error occured](https://gist.github.com/kashing1999/2706531444bc1207c31c6bed344c8af7)

I am also happy to help debug but I lack experience in these things. Thank you.

Mahooki, 2020-04-01:
I'm not super familiar with systemctl, but I wonder if this behavior points to it needing a different target. That is why I investigated starting it as a user service that is tied to the current session. I just couldn't figure out the correct permissions.

```text
...
[Install]
WantedBy=multi-user.target
```

Mahooki, 2020-04-07:
FWIW this behavior seems to only happen infrequently on Arch LTS, but happens every time on 5.6.2-arch1-2. I see there has been an update to restart on failure, I'm going to pull this update and attempt now.

PixlOne, 2020-04-07:
Could you try the latest commit? This bug seems related to #39 and I believe I've resolved that bug.

Mahooki, 2020-04-07:
Sure, I'll try now.

Mahooki, 2020-04-07:
That seems to have produced the desired behavior for me. It still fails with the `status=6/ABRT`, but it subsequently automatically restarts and functions properly. Worth noting that `yay -Syu` didn't seem to update the service file, so I just manually applied the changes that you made in the most recent commit.

For context, here is the journalctl -u logid output after a restart:
```text
-- Reboot --
Apr 07 19:24:33 archlinux systemd[1]: Started Logitech Configuration Daemon.
Apr 07 19:24:35 archlinux logid[560]: terminate called without an active exception
Apr 07 19:24:36 archlinux systemd[1]: logid.service: Main process exited, code=killed, status=6/ABRT
Apr 07 19:24:36 archlinux systemd[1]: logid.service: Failed with result 'signal'.
Apr 07 19:24:36 archlinux systemd[1]: logid.service: Scheduled restart job, restart counter is at 1.
Apr 07 19:24:36 archlinux systemd[1]: Stopped Logitech Configuration Daemon.
Apr 07 19:24:36 archlinux systemd[1]: Started Logitech Configuration Daemon.
Apr 07 19:24:36 archlinux logid[708]: [WARN] Failed to open /dev/hidraw15: open: No such file or directory
...
```

PixlOne, 2020-04-08:
Alright, thanks. I think it's safe to say this issue's resolved.

closed 2020-04-08 by PixlOne

1hirak, 2022-10-27:
I faced the similar issue on ubuntu 22.04 with MX Master 3

The following command worked for me

`sudo systemctl enable logid && sudo systemctl --user enable logid && sudo systemctl restart logid`