PixlOne/logiops #37: Key combinations don't seem to work

afnanenayet, 2020-04-01:
I have a section in my configuration that looks like this:

```
    {
        cid: 0xc3;

        action = 
        {
            type: "Gestures";
            gestures: (
                {
                    direction: "Left";
                    mode: "OnRelease";
                    action = 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFT", "KEY_LEFTSHIFT", "KEY_LEFTMETA"];
                    };
                },
                {
                    direction: "Right";
                    mode: "OnRelease";
                    action = 
                    {
                        type: "Keypress";
                        keys: ["KEY_RIGHT", "KEY_LEFTSHIFT", "KEY_LEFTMETA"];
                    };
                }
            );
        };
    }
```
Whenever I do these gestures, it seems like only the first key in the list is "pressed" by `logid`. Is this intended behavior? I saw that the wiki notes that the `keys` section can be a list of multiple keys.

PixlOne, 2020-04-01:
I can't seem to replicate that issue, try monitoring the logid device with evtest.

Mahooki, 2020-04-01:
I also was not able to replicate the issue - I have these settings in my config and they all seem to be working fine:
```text
devices: (
{
    name: "MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 20;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 600;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "NoPress";
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_H"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_L"];
                        }
                    },
                    {
                        direction: "None"
                        mode: "OnRelease"
                        action =
                        {
                            type: "Keypress";
                            keys: ["BTN_MIDDLE"];
                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_M"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "NoPress";
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_H"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_L"];
                        }
                    },
                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFTSHIFT", "KEY_S"];
                        };
                    }
                );
            };

        },
    );
}
);
```

afnanenayet, 2020-04-02:
Ah guys I figured it out, it turns out that the order of the keypresses does matter.

meta, shift, left/right worked

closed 2020-04-02 by afnanenayet