PixlOne/logiops #38: Feature: Action - run command

rhysperry111, 2020-04-03:
Hi, it would be nice if you allowed the user to specify custom commands. This would be useful for something like a gesture so that when you move your mouse up/down it will turn the brightness up/down

rhysperry111, 2020-04-03:
I will submit a PR soon to implement this, but I am relatively new to C/C++ so it might not be of the best quality

PixlOne, 2020-04-03:
I was going to do this but I ran into an issue. Whenever a command is run through this program, it would run it as the same user as logid. Considering many people run this service as root, this would mean all commands are run as root.

Of course, you could use sudo to run the program as another user, but this would mean the user to run as would be explicitly defined and some environment variables (particularly ones relating to the desktop session) would not be properly defined.

I'm currently working on an IPC implementation and one of the planned features is a way to communicate with a separate daemon that runs on the user's desktop session. This should allow me to avoid this issue entirely.

If you want, I could write a branch with a command action anyways. But at this point, it's probably better to use xbindkeys and use a Keypress action.

rhysperry111, 2020-04-03:
It would be nice to see this feature soon, but I can wait

2agr6v9u7mxu, 2020-07-22:
This would really open up the options for KDE users. 

I currently use fusuma for my trackpad configuration to talk directly to the qdbus commands available through the KDE API, for example:

`'qdbus org.kde.kglobalaccel /component/kwin org.kde.kglobalaccel.Component.invokeShortcut "Switch One Desktop to the Right"'`

This is really handy when trying swap between a virtual machine and the host for example, as it is a direct system call not a keypress that gets swallowed by the vm.

emilcarr, 2020-07-22:
A user-side daemon sounds unnecessarily complicated. Mapping an action to a mouse button keypress and catching that user-side with sxhkd or xbindkeys should work fine.

 Although maybe that won't be true with the interface pixl has mentioned developing.

2agr6v9u7mxu, 2020-07-22:
I agree. Hopefully, the interface solves this issue. Correct me if I'm wrong, but currently, there is no way to chain key bidings together? 

This is an issue when trying to escape a VM, you need to first press the hotkeys and then the key combination required for the host, hence, I use the system command mentioned above to avoid this issue with my trackpad config. I'm trying to achieve the same workflow with my mouse as well as the trackpad.