PixlOne/logiops #39: cannot autostart with systemctl

lorenzopompili, 2020-04-05:
Hello, 
if i run logiops with `systemctl start logid` it's all fine and work like a charm but if i enable it with `systemctl enable logid` and restart my pc, i have this issue:
```
● logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; enabled; vendor preset: disabled)
     Active: failed (Result: signal) since Sun 2020-04-05 11:31:27 CEST; 33s ago
    Process: 668 ExecStart=/usr/bin/logid (code=killed, signal=ABRT)
   Main PID: 668 (code=killed, signal=ABRT)
apr 05 11:31:19 myself-dell5570 systemd[1]: Started Logitech Configuration Daemon.
apr 05 11:31:26 myself-dell5570 logid[668]: [WARN] Failed to open /dev/hidraw5: open: No such file or directory
apr 05 11:31:26 myself-dell5570 logid[668]: terminate called without an active exception
apr 05 11:31:27 myself-dell5570 systemd[1]: logid.service: Main process exited, code=killed, status=6/ABRT
apr 05 11:31:27 myself-dell5570 systemd[1]: logid.service: Failed with result 'signal'.

```

How can i fix the autostart?
thanks

rhysperry111, 2020-04-06:
What kernel are you on? I just upgraded to 5.6 and am also experiencing this issue

rhysperry111, 2020-04-06:
Actually, just realized that it doesn't work on 5.6 no matter what I do (systemd or cmdline)

lorenzopompili, 2020-04-06:
It's working for me, i just have a problem with a reboot like I described before. I have to run logid manually only

Jannik2099, 2020-04-07:
I'm guessing it starts the service before the device is ready. Copy the service file to /etc/systemd/system , and add the following lines to the [Service] part of the file:
```
Restart=on-failure
StartLimitIntervalSec=0
```

lorenzopompili, 2020-04-07:
> I'm guessing it starts the service before the device is ready. Copy the service file to /etc/systemd/system , and add the following lines to the [Service] part of the file:
> 
> ```
> Restart=on-failure
> StartLimitIntervalSec=0
> ```

it's a nice solution and i was thinking about it. It's like a loop restart but if i decide to unplug the device? He's continuing the loop. Maybe a timer "at user login?"

PixlOne, 2020-04-07:
After reading the journalctl logs, I appear to be getting a broken pipe system error occassionally.

I'll try to fix this, but in the meantime I could add an auto-restart feature to the systemd service as a workaround.

PixlOne, 2020-04-07:
Alright @Jannik2099, I've added your code to the logid.service.cmake file in the repository.

It seems to have fixed this issue on my computer but I'll wait for further testing from others before I close this issue.

mwkemo, 2020-04-12:
I was getting error while checking dmesg after reboot. With new systemd version 230 and above,  StartLimitIntervalSec name is valid only in [Unit].

muesli, 2020-04-29:
Just a heads up: same here, occasionally getting a broken pipe.