PixlOne/logiops #41: /etc/logid.cfg not found

tirefrier, 2020-04-10:
----- $systemctl status logid --------
● logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; enabled; vendor preset: disabled)
     Active: active (running) since Fri 2020-04-10 10:13:27 CDT; 3s ago
   Main PID: 4447 (logid)
      Tasks: 3 (limit: 38408)
     Memory: 1.4M
     CGroup: /system.slice/logid.service
             └─4447 /usr/bin/logid

Apr 10 10:13:27 helios systemd[1]: Started Logitech Configuration Daemon.
Apr 10 10:13:27 helios logid[4447]: **[WARN] No button configuration found, reverting to null config.**

-------- $ ls /etc/ | grep logi -----------
logid.cfg
login.defs

-------- $ cat /etc/logid.cfg -----------
devices: (
{
    name: "MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F10"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F7"];
                        };
                    },
#                    {
#                        direction: "Left";
#                        mode: "OnRelease";
#                        action =
#                        {
#                            type: "CycleDPI";
#                            dpis: [50, 500, 1000, 1500, 2000, 3000, 4000];
#                        };
#                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFT"];
                        };
                    },

#                    {
#                        direction: "Right";
#                        mode: "OnRelease";
#                        action =
#                        {
#                            type = "ToggleHiresScroll";
#                        }
#                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_RIGHT"];
                        }
                    },

                    {
                        direction: "None"
                        mode: "OnRelease"
                        action = 
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F9"];
                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        # {
        #     # Next tab instead of fwd in history, Comment to default behavior
        #     cid: 0x53;
        #     action =
        #     {
        #         type :  "Keypress";
        #         keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
        #     };
        # },
        # {
        #     # Previous tab instead of back in history, Comment to default behavior
        #     cid: 0x56;
        #     action =
        #     {
        #         type :  "Keypress";
        #         keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
        #     };
        # }
    );
},
{
# Another device to configure
name: "Other Logitech USB Receiver";

}
);

PixlOne, 2020-04-10:
It appears to find the configuration file but it can't find any button mappings in the configuration for your device.

Do you have a buttons field in your configuration for the device?

Check out a sample config here: https://github.com/PixlOne/logiops/blob/master/logid.example.cfg

Also, be sure to check out [the wiki page on how to configure the device](https://github.com/PixlOne/logiops/wiki/Configuration)

tirefrier, 2020-04-10:
I do have a buttons field in my .cfg, I actually believe I am using your sample with only a few minor changes. I'll attach it below, I had to change the extension for it to allow me to upload it, but it is /etc/logid.cfg on my system.

[logid.cfg.txt](https://github.com/PixlOne/logiops/files/4463628/logid.cfg.txt)

PixlOne, 2020-04-11:
The error message is referring to "Other Logitech USB Receiver".

I'll try to remove the error message (in case a device doesn't support ReprogControls) and include line numbers/device names for the error messages.

The configuration on the MX Master 3 seems to be fine so I'll close the issue for now.

closed 2020-04-11 by PixlOne