PixlOne/logiops #43: Scrolling speed becomes super slow with logid active

muesli, 2020-04-12:
Since a recent update scrolling with the wheel is unusably slow, as soon as logid is running. Stopping the process alone doesn't work either, I actually need to turn my mouse off and on again to make it scroll normally again, too.

Using an MX Master 3.

PixlOne, 2020-04-12:
This is likely something to do with how Linux 5.6 introduced MX Master 3 support to the hid-logitech-hidpp module.

Try enabling hires scrolling, that should allow your scrolling to properly adjust to the new module.

muesli, 2020-04-12:
Oh, that actually makes a lot of sense! I probably updated both at the same time! Testing later tonight.

muesli, 2020-04-13:
Works fine as soon as I've enabled `hires`, indeed! Thank you!

closed 2020-04-13 by muesli