PixlOne/logiops #47: Controlling volume w/ thumb wheel?

wazlecracker, 2020-04-24:
How could one control volume with the horizontal thumb wheel? I've tried using CIDs 0x5b & 0x5d and mapping them to the keypresses KEY_VOLUMEDOWN & KEY_VOLUMEUP respectively, but nothing occurs.
In xbindkeys I use
```
# thumb wheel up => increase volume
"xte 'key XF86AudioRaiseVolume'"
   b:7

# thumb wheel down => lower volume
"xte 'key XF86AudioLowerVolume'"
   b:6
```
But I'd prefer to run all my mouse controls through a single program.

dhensen, 2020-06-11:
This issue is present. I was running `xev` and was rolling the thumbweel up/down ... I see no events at all. I can confirm that my xbindkeys is working properly by temporary assign another button (for instance when I use b:1 instead of b:7, I can see the volume raise.

I would happily fix this issue, any pointers on where to look in the code?

Elijas, 2020-06-11:
What mouse are you using?

These two work perfectly for me, Pop OS 20.04, Master MX 3
```
"pactl set-sink-volume @DEFAULT_SINK@ -1200"
  b:6
"pactl set-sink-volume @DEFAULT_SINK@ +1200"
  b:7
```
```
"amixer -D pulse sset Master 3%-"
  b:6
"amixer -D pulse sset Master 3%+"
  b:7
```

wazlecracker, 2020-06-12:
@Elijas those button assignments aren't from LogiOps.

dhensen, 2020-06-13:
@wazlecracker Then which program/driver is making those button assignments work on Pop OS 20.04???

I'm using MX Master 2S.

Elijas, 2020-06-16:
> @wazlecracker Then which program/driver is making those button assignments work on Pop OS 20.04???
> 
> I'm using MX Master 2S.

Does `xte 'key XF86AudioLowerVolume'` work from command line ?

Elijas, 2020-06-16:
> run all my mouse controls through a single program

As far as I remember, Logitech doesn't provide standard, simple way to intercept the thumb wheel, so unless Logiops would build additional layer of functionality on top, we'll have to use other software

wazlecracker, 2020-06-16:
@dhensen not sure on your computer, but those assignments are what xbindkeys uses on mine. Logiops uses names like "0x5b" for buttons.

@Elijas I'm not sure about the standardization, but obviously xbindkeys has it figured out. Perhaps crib from them?