PixlOne/logiops #48: MX Master name issues

levaidaniel, 2020-04-24:
Hi!

I'm new to this so this is most certainly an overlook on my part, but I'm trying to find out what should I put into my logid.cfg to configure my mouse.
It's an MX Master and connected with a Logitech _Unifying_ receiver to a USB hub. This is the output from logid:
```
[INFO] Missing hiresscroll option, not setting.
[INFO] Found Logitech USB Receiver on /dev/hidraw5
[DEBUG] Receiver on /dev/hidraw5: Device 1 paired
[DEBUG] Receiver on /dev/hidraw5: Device 2 paired
[DEBUG] Receiver on /dev/hidraw5: Device 2 paired
[INFO] Link lost to
[DEBUG] Receiver on /dev/hidraw5: Device 2 paired
[DEBUG] Receiver on /dev/hidraw5: Connection established to device 2
```
I switched it off/on to see if it reconnects.
Xinput tells me this:
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech MX Keys                          id=11   [slave  pointer  (2)]
⎜   ↳ Logitech Wireless Mouse PID:4071          id=13   [slave  pointer  (2)]
⎜   ↳ logid                                     id=7    [slave  pointer  (2)]
```
So I'm not really sure what do I have to put into logid.cfg. I've tried:
```
name: "MX Master";
name: "Logitech Wireless Mouse MX Master 3";
name: "Logitech Wireless Mouse";
name: "Logitech Wireless Mouse PID:4071";
name: "2";
name: "Logitech USB Receiver";
name: "*";
```

None of those seem to work :-\

PixlOne, 2020-04-25:
It should say `[INFO] <device name> detected: device <number> on <path>` in the logs when it detects a device. It says `[INFO] MX Master detected: device 1 on /dev/hidraw0` for me.

The device name should be "MX Master" but I'm not quite sure if the issue you're having is necessarily with naming the device.

pysiekytel, 2020-04-26:
I had some problems with name of device after upgrade to kernel `5.6`
Before that, name of my mouse was  `Logitech MX Master 3`
Now I have to use `Logitech Wireless Mouse MX Master 3` to get device properly configured
[image]

I know that there were many changes in 5.6 kernel done by Logitech: https://www.phoronix.com/scan.php?page=news_item&px=Logitech-Improvements-Linux-5.6

levaidaniel, 2020-04-26:
This is 5.6.4 here as well, and the weird thing is, that now that I've tried this countless times (look for device name in `logid -v` output -> edit logid.cfg -> restart, that is), it seems that sometimes it actually works, then right after that, if I turn off/on the mouse, it sometimes aborts:
```
$ sudo logid -v
[INFO] Missing hiresscroll option, not setting.
[INFO] Found Logitech USB Receiver on /dev/hidraw1
[DEBUG] Receiver on /dev/hidraw1: Device 1 paired
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
[INFO] Link lost to
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
[DEBUG] Receiver on /dev/hidraw1: Connection established to device 2
[INFO] Device MX Master not configured, using default config.
[INFO] MX Master detected: device 2 on /dev/hidraw1
[DEBUG] 0x800 was pressed but no action was found.
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
^C
$ sudo logid -v
[INFO] Found Logitech USB Receiver on /dev/hidraw1
[DEBUG] Receiver on /dev/hidraw1: Device 1 paired
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
[INFO] Link lost to
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
[DEBUG] Receiver on /dev/hidraw1: Connection established to device 2
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw3 (possibly asleep)
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
           [ turn mouse off/on]
[INFO] Link lost to
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Receiver on /dev/hidraw1: Device 2 paired
[DEBUG] Receiver on /dev/hidraw1: Connection established to device 2
terminate called after throwing an instance of 'std::runtime_error'
  what():  Invalid DevicePairingInfo type
Aborted

```

PixlOne, 2020-05-15:
You need it running in the background constantly.

I'm using an MX Master and I don't have that issue. What receiver device id/mouse device id are you using?

levaidaniel, 2020-05-18:
I've had a recent upgrade to kernel 5.6.10 (ARM), and now it seems to be stable enough! I'm using it via Bluetooth however, these days.
I need to have these in my _logid.cfg_, depending on using either the Unifying gadget or BT:
```
// Logitech Unifyer connection:
//name: "MX Master";

// Bluetooth connection:
name: "Logitech Wireless Mouse MX Master";
```
The mouse actually attaches to _logitech-hidpp_ now, instead of _hid-generic_, like it used to.

kgorol, 2020-06-13:
I am also having some issues with my MX Master  2S included in MX900 bundle. (https://www.amazon.com/Logitech-Performance-Premium-Backlit-Keyboard/dp/B07C6JCFY5)
I am also using Unified Receiver

```
[kamil@kamil-pc ~]$ sudo logid
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[kamil@kamil-pc ~]$ sudo logid -v
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[DEBUG] Receiver on /dev/hidraw0: Device 1 paired
[DEBUG] Receiver on /dev/hidraw0: Device 2 paired
[kamil@kamil-pc ~]$ sudo systemctl restart logid
[kamil@kamil-pc ~]$ sudo logid --verbose debug
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[DEBUG] Receiver on /dev/hidraw0: Device 1 paired
[DEBUG] Receiver on /dev/hidraw0: Device 2 paired
[DEBUG] Receiver on /dev/hidraw0: Device 1 paired
[INFO] Link lost to 
[DEBUG] Receiver on /dev/hidraw0: Device 1 paired
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
```