PixlOne/logiops #50: Unable to set DPI for MX Master

levaidaniel, 2020-04-27:
I'm getting these on Linux 5.6.4, with `git-2ed3efa`:
```
logid[3580996]: [ERROR] Error while setting DPI: Invalid argument
logid[3580996]: [ERROR] HID++ 1.0 Error whjle configuring MX Master: Resource error
logid[3580996]: [ERROR] Error while setting DPI: Invalid argument
logid[3580996]: [ERROR] Error while setting DPI: Invalid argument
```
But only in the systemd service's journal output. When running `logid -v` from console it doesn't say anything like this.

The mouse is connected with a unifying receiver.

PixlOne, 2020-05-15:
Can I see your logid configuration file?

levaidaniel, 2020-05-15:
Oops, sorry about this, I forgot to update.
It was actually a valid response from logid; I tried to set a DPI too high for my device. This a 1st gen MX Master and apparently it can only do 1500 tops.

closed 2020-05-15 by levaidaniel