PixlOne/logiops #54: Compile failure on Fedora 32

lucky-day, 2020-05-03:
Trying to compile logiops I get a build failure.

```
$ rm -rf build;mkdir build;cd build
[lucas@localhost build]$ cmake ..
-- The C compiler identification is GNU 10.0.1
-- The CXX compiler identification is GNU 10.0.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.9.0
-- Checking for module 'systemd'
--   Found systemd, version 245
-- Checking for module 'libconfig'
--   Found libconfig, version 1.7.2
Could not find libhidpp include dir, getting submodule
-- Checking for module 'libudev'
--   Found libudev, version 245
-- systemd units will be installed at /usr/lib/systemd/system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lucas/github/logiops/build
```
And then running make
```
$ make
Scanning dependencies of target hidpp
[  1%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/misc/Log.cpp.o
[  2%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/misc/CRC.cpp.o
[  4%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/RawDevice_linux.cpp.o
[  5%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/DeviceMonitor_linux.cpp.o
[  7%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/UsageStrings.cpp.o
[  8%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/Dispatcher.cpp.o
[ 10%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/SimpleDispatcher.cpp.o
[ 11%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/DispatcherThread.cpp.o
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In constructor ‘HIDPP::DispatcherThread::DispatcherThread(const char*)’:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:70:8: warning: suggest explicit braces to avoid ambiguous ‘else’ [-Wdangling-else]
   70 |     if (!checkReportDescriptor (rdesc))
      |        ^
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In member function ‘virtual void HIDPP::DispatcherThread::sendCommandWithoutResponse(HIDPP::Report&)’:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:108:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  108 |     if (forceLongReports() && report.type() == Report::Short)
      |     ^~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:110:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  110 |  _dev.writeReport (report.rawReport ());
      |  ^~~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In member function ‘virtual std::unique_ptr<HIDPP::Dispatcher::AsyncReport> HIDPP::DispatcherThread::sendCommand(HIDPP::Report&&)’:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:115:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  115 |     if (forceLongReports() && report.type() == Report::Short)
      |     ^~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:117:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  117 |  std::unique_lock<std::mutex> lock (_command_mutex);
      |  ^~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:121:77: warning: missing initializer for member ‘HIDPP::DispatcherThread::Command::response’ [-Wmissing-field-initializers]
  121 |  auto it = _commands.insert (_commands.end (), Command { std::move (report) });
      |                                                                             ^
[ 13%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/Device.cpp.o
In file included from /home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:19:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.h:46:8: error: ‘string’ in namespace ‘std’ does not name a type
   46 |   std::string _msg;
      |        ^~~~~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.h:25:1: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
   24 | #include <tuple>
  +++ |+#include <string>
   25 | 
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.h:83:7: error: ‘string’ in namespace ‘std’ does not name a type
   83 |  std::string name () const;
      |       ^~~~~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.h:83:2: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
   83 |  std::string name () const;
      |  ^~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.h:105:7: error: ‘string’ in namespace ‘std’ does not name a type
  105 |  std::string _name;
      |       ^~~~~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.h:105:2: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
  105 |  std::string _name;
      |  ^~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp: In constructor ‘HIDPP::Device::InvalidProtocolVersion::InvalidProtocolVersion(const std::tuple<unsigned int, unsigned int>&)’:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:38:2: error: ‘_msg’ was not declared in this scope
   38 |  _msg = ss.str ();
      |  ^~~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp: In member function ‘virtual const char* HIDPP::Device::InvalidProtocolVersion::what() const’:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:43:9: error: ‘_msg’ was not declared in this scope
   43 |  return _msg.c_str ();
      |         ^~~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp: In constructor ‘HIDPP::Device::Device(HIDPP::Dispatcher*, HIDPP::DeviceIndex)’:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:61:4: error: ‘_name’ was not declared in this scope; did you mean ‘rename’?
   61 |    _name = ireceiver.getDeviceName (device_index - 1);
      |    ^~~~~
      |    rename
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:75:3: error: ‘_name’ was not declared in this scope; did you mean ‘rename’?
   75 |   _name = _dispatcher->name ();
      |   ^~~~~
      |   rename
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp: At global scope:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:113:13: error: no declaration matches ‘std::string HIDPP::Device::name() const’
  113 | std::string Device::name () const
      |             ^~~~~~
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:113:13: note: no functions named ‘std::string HIDPP::Device::name() const’
In file included from /home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.cpp:19:
/home/lucas/github/logiops/src/logid/hidpp/src/libhidpp/hidpp/Device.h:36:7: note: ‘class HIDPP::Device’ defined here
   36 | class Device
      |       ^~~~~~
make[2]: *** [src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/build.make:187: src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/Device.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:161: src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/all] Error 2
make: *** [Makefile:150: all] Error 2
```

Friedrich42, 2020-05-09:
```
dnf install -y libevdev systemd-devel libconfig libevdev-devel libconfig-devel
echo /usr/local/lib >> /etc/ld.so.conf.d/lidhidpp.conf
ldconfig

```

Friedrich42, 2020-05-09:
```
dnf install -y libevdev systemd-devel libconfig libevdev-devel libconfig-devel
echo /usr/local/lib >> /etc/ld.so.conf.d/lidhidpp.conf
ldconfig
```
I tried to install these libraries, but I have the same issue when compiling

Friedrich42, 2020-05-09:
Actually there is a workaround. You can install fedora:31 in boxes app, compile it in container and then just copy to your local machine and make install it

PixlOne, 2020-05-15:
Oh, that's a tedious workaround but it works for now.

I'll see what I can do to solve this but it seems to be unable to compile the entirety of libhidpp.

I think I may only need to add a single `#include <string>` statement in hidpp/src/libhidpp/Device.h, can someone try that out and test it for me? It saves me the trouble of installing Fedora 32 and testing it on there.

Thanks.

PixlOne, 2020-05-15:
This should be fixed on the latest commit, test it out and let me know.

Friedrich42, 2020-05-31:
Sorry for late answer. Issue can be closed, all works ok on fc32, tested just now. Thank you!

closed 2020-05-31 by PixlOne