PixlOne/logiops #55: Compile failure on Ubuntu 18.04 LTS

ITEFARMAT, 2020-05-07:
Hello could you please, handle with it ?

-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)

TomasDePomas, 2020-05-08:
I am having this same issue, same ubuntu version, exact same error.

TomasDePomas, 2020-05-08:
Ah, found the solution, we seem to be missing a couple dependencies, this solves this error:
```apt-get install pkg-config```

ITEFARMAT, 2020-05-08:
@TomasDePomas  Thanks , it was helpful but not completly. 
Stil I recive problems:

-- Checking for module 'libconfig'
--   No package 'libconfig' found
Could not find libhidpp include dir, getting submodule
-- Checking for module 'libudev'
--   No package 'libudev' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  src/logid/CMakeLists.txt:57 (pkg_check_modules)

I downloaded libconfig from :+1: 
https://ubuntu.pkgs.org/18.04/ubuntu-universe-arm64/libconfig-dev_1.5-0.4_arm64.deb.html

And I can't still compile it ...

PixlOne, 2020-05-08:
You need to install libudev-dev

ITEFARMAT, 2020-05-13:
@PixlOne 

One more problem below:

fatal error: libconfig.h++: There is no such file or directory
 #include <libconfig.h++>
          ^~~~~~~~~~~~~~~
compilation terminated.
src/logid/CMakeFiles/logid.dir/build.make:62: recipe for target 'src/logid/CMakeFiles/logid.dir/logid.cpp.o' failed
make[2]: *** [src/logid/CMakeFiles/logid.dir/logid.cpp.o] Error 1
CMakeFiles/Makefile2:87: recipe for target 'src/logid/CMakeFiles/logid.dir/all' failed
make[1]: *** [src/logid/CMakeFiles/logid.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

alejgarciarodriguez, 2020-05-14:
> @PixlOne
>
> One more problem below:
>
> fatal error: libconfig.h++: There is no such file or directory
> #include <libconfig.h++>
> ^~~~~~~~~~~~~~~
> compilation terminated. [...]

Same problem here, I have libconfig-dev 1.5-0.4 in Ubuntu 18.04

alejgarciarodriguez, 2020-05-14:
I was able to compile it by installing this:

`sudo apt-get install libconfig++-dev`