PixlOne/logiops #56: Could not find libhidpp include dir...

Aru1199, 2020-05-08:
I'm trying to compile the program, but I'm getting this error. I've browsed the other issues here on Github, but I can't seem to find a solution. `sudo ln -s /usr/local/lib/libhidpp.so /usr/lib/libhidpp.so` did not work.

```
Could not find libhidpp include dir, getting submodule
fatal: not a git repository (or any of the parent directories): .git
CMake Error at src/logid/CMakeLists.txt:65 (add_subdirectory):
  add_subdirectory given source "hidpp/src/libhidpp" which is not an existing
  directory.

-- systemd units will be installed at /lib/systemd/system
-- Configuring incomplete, errors occurred!

```

PixlOne, 2020-05-08:
Oh, did you download the repo as a zip file? The repository relies on a submodule, try downloading this repo using `git clone https://github.com/PixlOne/logiops.git` and then compiling it.

Aru1199, 2020-05-08:
Yes I was trying to compile using the zip file. I was able to install the program successfully after cloning the repository. Thank you!

closed 2020-05-08 by Aru1199