PixlOne/logiops #57: Can't log in to Ubuntu

Aru1199, 2020-05-08:
Hi, I just installed logiops, ran the command  `sudo systemctl start logid`, and rebooted for the first time since the installation. Now the screen would be stuck at the purple screen, with the cursor frozen in place. Logiops is the only program I installed/updated before restarting. Have there been any examples of this before, and/or how can I fix this?

Aru1199, 2020-05-08:
I couldn't fix this even after a reinstall of Ubuntu, so I decided to do a clean installation instead. It would still be nice to get some insight into this issue, since I'm now afraid to install this app.

PixlOne, 2020-05-08:
That's a strange issue, I'll see if I can replicate it. What version of Ubuntu are you using?

Aru1199, 2020-05-09:
I'm using Ubuntu 20.04.