PixlOne/logiops #58: Option only for one application

andreacocci95, 2020-05-09:
I setted MX Master 3 with your program and it works fine! The only thing that I can't find is the possibility to add specific options for specific application. Is it possible?

Elijas, 2020-05-13:
You could have this by making your MX Master just click some long hotkey such as "Ctrl Alt Shift F13" and then using other more advanced hotkey software which can recognize active applications and run different actions based on the "Ctrl Alt Shift F13" hotkey. I used Autohotkey on windows to do exactly this, maybe xbindkeys could be an alternative on linux

PixlOne, 2020-05-15:
I'll look into this after I've fully implemented IPC in logid.

Elijas, 2020-05-27:
Another workaround could be having a daemon running in background, which monitors the currently active aplication and reloads the _logid_ with the appropriate cfg for that application

MattHag, 2020-10-03:
That would be a really great feature and gives the opportunity to create more advanced configurations to improve usability.

It would also help to create fine-grained default configuration for several programs, as it is done with each MX Master 3 using Logitech Options.

dasboot1, 2021-11-25:
Would love to have that feature 👍

eugenecherepanov, 2022-05-26:
Hi! Do you have any news about this feature?

P.S. Logiops great app many thanks for your job

GarThor, 2025-02-09:
Can I second this feature?

I'd really like to apply different shortcuts across different applications, without having to run a second or third program.