PixlOne/logiops #59: Turn OFF/ON my mouse to be recognized

mwkemo, 2020-05-10:
Mouse: MX Master 3
Linux: Arch
Kernel: 5.6.11
DE: CInnamon

logid service is enabled and it starts at system boot without errors. After each boot I always have to manually turn OFF and back ON my mouse for logid to recognize it. Sometimes even ON/OFF doesn't work so I have to manually restart logid service and turn OFF/ON my mouse to be recognized by logid. I have tried with AUR package "logiops-git" and manually compiling from git but it seams there is no difference.
When I run "journalctl -u logid" I get a lot of messages:

`logid[671315]: [DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)`
`logid[671315]: [DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)`

This is my hidpp-list-devices output:

`/dev/hidraw0: Logitech USB Receiver (046d:c52b) HID++ 1.0`
`/dev/hidraw0 (device 1): K800 (046d:2010) HID++ 1.0`
`/dev/hidraw0 (device 2): MX Master 3 (046d:4082) HID++ 4.5`

Any help is appreciated.

mwkemo, 2020-05-10:
Now i have tried with two Unifying Receivers. One for keyboard and for for my mouse. After checking log file I get message:

`logid[588]: terminate called after throwing an instance of 'std::system_error'`
`logid[588]:   what():  write: Broken pipe`

It seams this is related with #39

PixlOne, 2020-05-11:
That's odd, is the behaviour any different when you only have the mouse paired?

schuppentier, 2020-05-11:
I have the exact same Problem on Arch with my MX Master 3, but I have no keyboard paired, only the mouse.

mwkemo, 2020-05-11:
Same log message with mouse only paired without keyboard just as @schuppentier comment.

jef, 2020-05-20:
Same here. Also have the `MX Keys`. I put in empty block for the `MX Keys` to see if the config would bypass it, but no avail.

jfk88, 2020-07-09:
I'm afraid same here. What happens in my case is random _detection_. It works for a while and suddenly `MX Master 3` stop getting detected (but never `MX Keys`). `Device 1` is paired but not detected:

```
$ sudo logid -v
[INFO] Found Logitech USB Receiver on /dev/hidraw2
[DEBUG] Receiver on /dev/hidraw2: Device 1 paired
[DEBUG] Receiver on /dev/hidraw2: Device 2 paired
[INFO] Device MX Keys not configured, using default config.
[INFO] MX Keys detected: device 2 on /dev/hidraw2
```
Is anyone in contact with someone in the Solaar team? That might be helpful I've never been disconnected like this with Solaar.

PixlOne, 2020-07-09:
@jfk88 I'm working on a refactor of the program, could you test if the same issue occurs with the drop-libhidpp branch? The branch isn't complete but I'm using a different system for scanning devices (similar to how solaar and hid-logitech-dj does it).

jfk88, 2020-07-10:
@PixlOne I got this:

```
$ sudo logid -v
[INFO] Detected receiver at /dev/hidraw2
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw2:1
[INFO] Device MX Keys Wireless Keyboard not configured, using default config.
[INFO] Device found: MX Keys Wireless Keyboard on /dev/hidraw2:2
[ERROR] Failed to add device 1 to receiver on /dev/hidraw2: gestures
```

PixlOne, 2020-07-10:
Oh right, gestures aren't implemented yet. Run it with no config file by running `sudo logid -v -c /dev/null`

jfk88, 2020-07-13:
After a couple days of testing I'm glad to report that the `drop-libhidpp` branch greatly improve the detection of the devices. I had to reload the daemon just once after 5 hours of sleep time. For the record I've been using a config with one `KEYPRESS` function on the thumb button of MX Master 3. 

The only bug I've been encountering is at startups with systemd where the daemon failed in ACTIVE and SUB (though loaded) leading to that error:

`[ERROR] Error adding device /dev/hidraw2: Invalid sub ID`

This happens roughly 1 out of 3 startups. 

Looking forward to using it with full functionalities.

PixlOne, 2020-07-13:
Try the refactor branch, it's more up to date and based on master instead of feature/logictl. I've almost completed that branch and there's just a few bug fixes remaining on there. @jfk88

jfk88, 2020-07-14:
@PixlOne Indeed, flawless for now. Thank you for the great work.