PixlOne/logiops #60: Can't use gestures with Right Mouse Button (MX Master 3)

Elijas, 2020-05-13:
CID for right mouse button is 0x0051, a.k.a. 0x51 (source https://github.com/PixlOne/logiops/wiki/CIDs )
And can't use the gestures with it, although gestures with middle mouse button work file
 Code:
```
// Location: /etc/logid.cfg 
 devices: ({
  name: "MX Master 3";
  smartshift: { on: true; threshold: 50; };
  hiresscroll: { hires: false; invert: false; target: false; };
  dpi: 1500; // max=4000
  buttons: (
    // Top button
    { cid: 0x51; action = { type: "Gestures"; gestures: (
        { direction: "None";    mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_ENTER" ];}},
        { direction: "Up";      mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_LEFTCTRL", "KEY_INSERT" ];}},
        { direction: "Down";    mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_LEFTSHIFT", "KEY_INSERT" ];}},
        { direction: "Left";    mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_PLAYPAUSE" ];}},
        { direction: "Right";   mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_NEXTSONG" ];}}
    );};},
    // Middle mouse button
    { cid: 0x52; action = { type: "Gestures"; gestures: (
        { direction: "None";    mode: "OnRelease"; action = { type: "Keypress";         keys: [ "BTN_MIDDLE" ];}},
        { direction: "Up";      mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_F11" ];}},
        { direction: "Down";    mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_LEFTMETA", "KEY_M" ];}},
        { direction: "Left";    mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_LEFT" ];}},
        { direction: "Right";   mode: "OnRelease"; action = { type: "Keypress";         keys: [ "KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_RIGHT" ];}}
    );};}
  );
});
```

PixlOne, 2020-05-15:
The left and right mouse buttons cannot be diverted. This is the output of hidpp20-reprog-controls on my MX Master:

```
Control Task    Info    Fn      Capabilities    Group   Remap to groups
0x0050  0x0038  mouse   -       -       1       1
0x0051  0x0039  mouse   -       -       1       1
0x0052  0x003a  mouse   -       reprog, divert  3       1+2+3
0x0053  0x003c  mouse   -       reprog, divert  2       1+2
0x0056  0x003e  mouse   -       reprog, divert  2       1+2
0x00c3  0x00a9  mouse   -       reprog, divert  3       1+2+3
0x00c4  0x009d  mouse   -       reprog, divert  3       1+2+3
0x00d7  0x00b4  -       -       divert, virtual 4       -
```

I would imagine this behaviour is similar on other devices. Since this is part of the mouse's design, I can't really fix this using native hardware features. Solving this issue would require that I implement my own button diversion mechanism.

I could do this in the future, but as of now, I won't fix this, sorry.

Elijas, 2020-05-15:
> The left and right mouse buttons cannot be diverted. This is the output of hidpp20-reprog-controls on my MX Master:
>
> ```
> Control Task    Info    Fn      Capabilities    Group   Remap to groups [...]

No problem, I suspected this exact situation to be the case:)