PixlOne/logiops #61: Add Donate button to the project

Elijas, 2020-05-13:
I'd like to support this project - can we have a '.github/FUNDING.yml' file added to this repository (with, for example https://ko-fi.com/ or any other provider)?

Also, maybe we could consider adding a way (e.g. https://issuehunt.io/ ) to attract developer attention to bugs/issues? E.g. https://github.com/PixlOne/logiops/issues/60

PixlOne, 2020-05-15:
I'll consider this in the future.

I may also include a note that I'm looking for contributors as well.

fstreun, 2020-05-18:
@PixlOne: I'd be interested in contributing to this project. (I just don't know how to contact you)

PixlOne, 2020-05-19:
Yeah, I probably should have included my contact information, I can be reached at pixlone@protonmail.com. I've added that to my bio.

PixlOne, 2020-06-16:
I've created a donate button in the project README.md.

closed 2020-06-16 by PixlOne