PixlOne/logiops #63: Unable to detect config file (/etc/logid.cfg)

jackhhchan, 2020-05-19:
Hi,

I have set up the /etc/logid.cfg file and set the value of 'name' to "MX Master 3" as shown by starting logid.

However, it seems /etc/logid.cfg isn't detected. 

Note: I have tried to run logid earlier and /etc/logid.id is detected and run as expected. It then stopped detecting the file suddenly and haven't since.

I have tried:
- specifically setting path to /etc/logid.cfg using -c option.
- reinstalling logiops

[image]

Also, what's the (input-event-code)[https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h] for the default thumb button to show windows? (i.e. ctrl-alt-down arrow for linux mint)

Thanks heaps!

PixlOne, 2020-05-19:
Can you post your config file?

Also, I'm not quite sure what you mean by default thumb button, do you mean how to do Ctrl, alt, Down arrow? In that case just set the keys to `["KEY_LEFTCTRL", "KEY_LEFTALT", 'KEY_DOWN"]`.

ElijahLynn, 2020-06-01:
I am having this same issue after a default install of https://aur.archlinux.org/packages/logiops-git/ that installed a default /etc/logid.cfg file. I'll post back once I get it working. Maybe my /etc/logid.cfg was from a previous attempt as I did have Logiops installed a year ago trying to get a Logitech touchpad installed.

ElijahLynn, 2020-06-01:
K, got the file loading now, it always was loading but somehow the file laid down on /etc/logid.cfg had the name `"MX Master"` and needed to be `"MX Master 3"` for it to get picked up. Also worth noting is that when I do a `systemctl reload logid` the changes still took about 2-5 seconds to get picked up after the reload finished.

closed 2024-06-16 by jackhhchan