PixlOne/logiops #65: MX Ergo compatibility report

slavfox, 2020-06-05:
Just got an MX Ergo to test with logiops:

* DPI adjustment is not possible. If the dpi section is present at all in the logid config, it fails with:
```
[INFO] MX Ergo detected: device 1 on /dev/hidraw7
terminate called after throwing an instance of 'HIDPP20::UnsupportedFeature'
  what():  Feature [0x2201] AdjustableDPI unsupported.
fish: “logid -v” terminated by signal SIGABRT (Abort)
```
* Binding actions works. CIDs are as expected:
  * `0xed` for the DPI button
  * `0x5b` for scroll left
  * `0x5d` for scroll right
  * `0x56` for forward
  * `0x53` for back

thetechgeekster, 2020-10-07:
Hi! Can you please share your config file as a template for me?

slavfox, 2020-10-07:
Here's the relevant part:
```
devices =
(
    {
        name = "MX Ergo";
        buttons =
        (
            {
                cid = 91;
                action =
                {
                    type = "Keypress";
                    keys =
                    [
                        "KEY_BACK"
                    ];
                };
            },
            {
                cid = 93;
                action =
                {
                    type = "Keypress";
                    keys =
                    [
                        "KEY_FORWARD"
                    ];
                };
            }
        );
    }
);
```

TobiasDev, 2020-11-08:
@slavfox  how did you get your MX Ergo to be recongnized? I keep getting this:

`[INFO] Device MX Ergo Multi-Device Trackball  not configured, using default config.
[INFO] Device found: MX Ergo Multi-Device Trackball  on /dev/hidraw0:1`

Even though I have added the config file and used that exact name.

If you don't mind me asking. :)

TobiasDev, 2020-11-08:
Ah, got it working. I had to add a "space" after the name for some weird reason. "MX Ergo Multi-Device Trackball " like that. Now it works!

osbama, 2021-01-30:
I am using a rather big screen, and not being able to adjust DPI is a rather significant issue for me. Are you aware of any updates on this regard?

fk1blow, 2021-02-02:
@TobiasDev can you please post the whole config file? I wasn't able to change the dpi.

thank you

TobiasDev, 2021-02-02:
I never changed the DPI in my config. :( However, if you want to have a look at it, it lives here: https://github.com/TobiasDev/dot-files/blob/master/logiops/logid.cfg

mellolucas, 2021-11-22:
Thank you! I had noticed that space, but thought that couldn't be the problem.

ExposedCat, 2022-08-02:
Did anyone managed to rebind "Switch Devices" and/or "Sniper Mode" buttons? Debug table shows them as "reprog? True" but looks like it does not work

devmessias, 2022-08-26:
Hi @TobiasDev @fk1blow  have you discoverd how to change the DPI?

TobiasDev, 2022-08-30:
> Hi @TobiasDev @fk1blow have you discoverd how to change the DPI?

Sorry! I had to switch to Windows again because of work so I'm sadly not sure anymore about this. :(

osbama, 2022-08-31:
Hi,
DPI selection works perfectly for me for a while now. 
However, the mouse button clicks are still "sticky" i.e. Left mouse button takes too long to release, or sometimes does not release at all. This sometimes results in unwanted dragging and unregistered double clicks. Generally annoying, but since I come from an era where mouse used to have only one button and a ball at the bottom,  I managed to get used to it. 

This is not a hardware issue, since I also use the same device with a MAC and Windows.