PixlOne/logiops #66: Settings lost when switching devices

ightnapovial, 2020-06-08:
When changing devices on the MX Master 3 (pressing the device switch button on the mouse itself to change computers) gesture/ scroll settings are lost until logid is restarted. Running logid manually with -v shows no new lines printed when inputs on the mouse are changed. All lines shown below are from the initial startup:
`[INFO] Logitech Wireless Mouse MX Master 3 detected: device 255 on /dev/hidraw4
[INFO] Device MX Keys not configured, using default config.
[INFO] MX Keys detected: device 255 on /dev/hidraw5
[DEBUG] 0x800 was pressed but no action was found.
[INFO] Logitech Wireless Mouse MX Master 3 detected: device 0 on /dev/hidraw4
[DEBUG] 0x800 was pressed but no action was found.
[INFO] Device MX Keys not configured, using default config.
[INFO] MX Keys detected: device 0 on /dev/hidraw5
[DEBUG] 0x1800 was pressed but no action was found.`

This issue only appears on Debian testing with the mouse connected via bluetooth. On kubuntu with the usb receiver switching inputs has no effect.

PixlOne, 2020-06-09:
Logitech mice connected over bluetooth have a different name for some reason; `Logitech Wireless Mouse ` is added to the beginning of the name. Try adding a device entry with the name `Logitech Wireless Mouse MX Master 3`.

ightnapovial, 2020-06-09:
I actually had that as the device name in logid.cfg already.

PixlOne, 2020-06-09:
I'm a little confused on what your issue is? Are you saying that when you press the device switch button and switch to Bluetooth, logid does not recognise it?

ightnapovial, 2020-06-09:
Sorry, I should have explained it better.

When logid is run on pc 1 while the mouse is connected with bluetooth logid will see the mouse and apply my settings as intended. The issue is when I switch to pc 2 then back to 1, my settings are no longer applied and I have to restart logid in order to get my settings back again.

Settings affected are gesture and hires scrolling, basically both revert to default as if I didn't have a logid.cfg file.

ightnapovial, 2020-06-15:
This is not solved. The issues is not related to restarting the PC, it is related to pressing the switch device button on the mouse itself.

lambdageek, 2020-07-19:
I have the same issue with a "M720 Triathlon Multi-Device Mouse" that I connect over Bluetooth.  If i start `logid` first, then connect the mouse, the settings are not applied.  If I connect the mouse first, then run `logid -v` I see
```
INFO] Device found: M720 Triathlon Multi-Device Mouse on /dev/hidraw1:255
[DEBUG] /dev/hidraw1:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0x5b | YES     |         | YES        | YES
[DEBUG] 0x5d | YES     |         | YES        | YES
[DEBUG] 0xd0 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```
After that if I either turn off the mouse, or hit the device switch button and switch to another PC and then back to the original PC (still running `logid`), the settings from the cfg file are not used (actions aren't applied, etc).
There is no additional output fromg `logid`

It seems like `logid` does not get notified when devices are added or removed?

PixlOne, 2020-07-19:
@lambdageek Can you compile this: https://github.com/PixlOne/hidpp, run `sudo hidpp-list-features /dev/hidraw1`, and post the output? I think I might have a clue what's going on.

lambdageek, 2020-07-19:
@PixlOne sure

```
M720 Triathlon (046d:b015) is a HID++ 4.5 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x1d4b] Wireless device status
Feature 0x05: [0x0020] Reset
Feature 0x06: [0x0021] Crypto Identifier
Feature 0x07: [0x0007] Device Friendly Name
Feature 0x08: [0x1000] Battery status
Feature 0x09: [0x1814] Change host
Feature 0x0a: [0x1815] Hosts info
Feature 0x0b: [0x1b04] Reprog controls v4
Feature 0x0c: [0x1c00] ?
Feature 0x0d: [0x2205] Pointer speed
Feature 0x0e: [0x2100] Vertical scrolling
Feature 0x0f: [0x00c1] DFUcontrol 2
Feature 0x10: [0x00c2] DFUcontrol 3
Feature 0x11: [0x1803] ? (hidden, internal)
Feature 0x12: [0x1813] ? (hidden, internal)
Feature 0x13: [0x1830] ? (hidden, internal)
Feature 0x14: [0x18a1] ? (hidden, internal)
Feature 0x15: [0x1df3] ? (hidden, internal)
Feature 0x16: [0x1e00] Enable hidden features (hidden)
Feature 0x17: [0x1eb0] ? (hidden, internal)
Feature 0x18: [0x1861] ? (hidden, internal)
Feature 0x19: [0x18b1] ? (hidden, internal)
Feature 0x1a: [0x1850] ? (hidden, internal)
Feature 0x1b: [0x1e22] ?
Feature 0x1c: [0x1f03] ? (hidden, internal)
Feature 0x1d: [0x18c0] ? (hidden, internal)
Feature 0x1e: [0x2121] Hi-res wheel
```

PixlOne, 2020-07-19:
So what I believe is happening is that your mouse stays connected over Bluetooth, but reports wakeup events via `Feature 0x04: [0x1d4b] Wireless device status`. I'll try to apply a patch to handle those events soon and hopefully that will fix it.

Just one thing, does this only happen over Bluetooth?

ightnapovial, 2020-07-19:
Recently I've noticed it happening on the USB receiver as well as BT. Havent noticed any issues on Kubuntu still, only Debian.

PixlOne, 2020-07-19:
Recently on the receiver? Is this on the latest version?

ightnapovial, 2020-07-19:
Latest mouse firmware, not latest logiops. Is there a quick way to update or do you have to uninstall + reinstall?

lambdageek, 2020-07-19:
@PixlOne for me it only happens with BT.  Here's what happens when I used the USB receiver:

USB receiver plugged in:
```
[INFO] Detected receiver at /dev/hidraw2
[WARN] Error adding device /dev/hidraw3: std::exception
[INFO] Device found: M720 Triathlon Multi-Device Mouse on /dev/hidraw2:1
[DEBUG] /dev/hidraw2:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0x5b | YES     |         | YES        | YES
[DEBUG] 0x5d | YES     |         | YES        | YES
[DEBUG] 0xd0 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```

USB receiver unplugged:
```
[WARN] Error ignored on detached thread/task: _readReport read failed: Input/output error
[INFO] Receiver on /dev/hidraw2 disconnected
```

(I'm building 0928b39e034d73667026fae75ce5b6428f646195 from source on Debian on amd64, btw)

PixlOne, 2020-07-19:
@ightnapovial Yeah, just pull the latest version, compile it, and install it.

ightnapovial, 2020-07-19:
OK, thanks. I'll update and retest when I get home.

lambdageek, 2020-07-19:
> So what I believe is happening is that your mouse stays connected over Bluetooth, but reports wakeup events via Feature 0x04: [0x1d4b] Wireless device status.

I think you're onto something.  Here's what happens if I'm connected over BT, and then click "Turn off Bluetooth" in the Gnome settings panel

```
[WARN] Error ignored on detached thread/task: _readReport read failed: Input/output error
```

and then turn BT back on

```
[INFO] Device found: M720 Triathlon Multi-Device Mouse on /dev/hidraw3:255
[DEBUG] /dev/hidraw3:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0x5b | YES     |         | YES        | YES
[DEBUG] 0x5d | YES     |         | YES        | YES
[DEBUG] 0xd0 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```

so it's the same thing as the USB receiver unplug/plug.

The only time that the daemon doesn't notice the mouse disconnected/reconnected is:
1. if when I use the device switch button on the mouse; or,
2. if I use the power switch on the bottom of the mouse to turn it off and back on.

PixlOne, 2020-07-19:
Try it now, I pushed a commit that should fix it. @lambdageek @ightnapovial

lambdageek, 2020-07-19:
@PixlOne works for me. :tada: 
```
[INFO] /dev/hidraw3:255 woke up.
[INFO] /dev/hidraw3:255 woke up.
```
That's after switching devices, and after turning the power switch off and on.

closed 2020-07-19 by PixlOne