PixlOne/logiops #67: Need to restart logid at each startup

CorradoLanera, 2020-06-10:
This is possibly linked with #39, #59, and #66.

System: Debian 10 non-free + KDE / Linux 4.19.0
Mouse: MX Master

At each system restart (not after logout/login!) I have to manually
```
$ sudo systemctl restart logid
```

I am quite new to Linux, hence is it possible that I have configured something wrong or missed to configure.

What additional information can I provide to debug/fix it? (please, include specific command line I need to execute for you step by step: as I said, I hope not to be a complete newbe anymore but I am not completely confident yet, for sure :-))

Thank you very much (for this great project too!)

CorradoLanera, 2020-06-11:
```
$ sudo systemctl status logid
[sudo] password for corrado: 
● logid.service - Logitech Configuration Daemon
   Loaded: loaded (/lib/systemd/system/logid.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
$ 
```

PixlOne, 2020-06-11:
`Loaded: loaded (/lib/systemd/system/logid.service; disabled; vendor preset: enabled)`

The service seems to be disabled, can you try running `sudo systemctl enable logid`?

CorradoLanera, 2020-06-12:
```
$ sudo systemctl enable logid
[sudo] password for corrado:
Created symlink /etc/systemd/system/multi-user.target.wants/logid.service → /lib/systemd/system/logid.service.
```

Restarted, and it works! Great!!

For me to learn and possibly for others, in your opinion what can I did wrongly or miss to do in my first configuration/workflow?

closed 2020-06-13 by PixlOne