PixlOne/logiops #69: Gesture button chords?

jimduchek, 2020-06-11:
Just got an MX Master 3 today.  On my previous MX Performance, the 'gesture' button appeared as a separate 'normal' mouse button (just like any other) and I had an xbindkeys setup that allowed me to do 'chords' -- hold the gesture button then click the left or right button to do stuff.   Kinda got used to it. 
 Would such a thing be possible here?  Happy to dig into the code and and possibly do a PR if the answer is "yes" or "I don't know", but if you happen to know it's not possible I can save myself the effort.

rosshadden, 2020-07-16:
Yes you can map the gesture button to a "normal" button/key. This is what I do to get chords working, since they aren't supported first-class by logiops or [piper](https://github.com/libratbag/piper):

```
# logid.cfg
{ cid: 0xc3; action = { type: "Keypress"; keys: ["KEY_HANJA"];    }; }

# .Xmodmap
keycode 131 = Hyper_L
remove Mod4 = Hyper_L
add Mod3 = Hyper_L

# i3/config
bindsym --whole-window Mod3+button1 exec "~/dotfiles/bin/config volume !"
bindsym --whole-window Mod3+button3 exec "~/dotfiles/bin/config mic !"
bindsym --whole-window Mod3+button4 exec "~/dotfiles/bin/config volume +5"
bindsym --whole-window Mod3+button5 exec "~/dotfiles/bin/config volume -5"
```

This makes the gesture button register as `HYPER_L`, and I then map the following (abstracted by some script I wrote):
- `hyper+leftclick`: toggle mute
- `hyper+rightclick`: toggle mic mute
- `hyper+mousewheel`: change my volume

It's... not great. Mostly because the scroll isn't intercepted, it just scrolls and also changes my volume. The clicks are intercepted though. The scroll problem seems to be an i3 behavior thing, so it might even work perfectly if you adapt this to your own environment and use something different.

rosshadden, 2020-07-16:
The above answers the question in your title. To answer the description in your title more specifically, the `KEY_HANJA` is the "normal" key in this case. You can make it things like `KEY_A` or `BTN_LEFT` for example, but I skipped a step and explained my workflow for chording based on your question.

I actually didn't have this setup until I read this issue you posted which gave me the hint of using xbindkeys (I used xmodmap instead, but same idea). So ironically you helped me help you :D. Thanks!