PixlOne/logiops #71: Problems during installation

0xfederama, 2020-06-13:
I made `mkdir build`, `cd build`, `cmake ..`.
Here is the log
```-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.9.0
-- Checking for module 'systemd'
--   Found systemd, version 245
-- Checking for module 'libconfig'
--   Found libconfig, version 1.5
Could not find libhidpp include dir, getting submodule
fatal: not a git repository (or any of the parent directories): .git
-- Checking for module 'libudev'
--   Found libudev, version 245
CMake Error at src/logid/CMakeLists.txt:65 (add_subdirectory):
  add_subdirectory given source "hidpp/src/libhidpp" which is not an existing
  directory.

-- systemd units will be installed at /lib/systemd/system
-- Configuring incomplete, errors occurred!
See also "/home/federico/Downloads/logiops-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/federico/Downloads/logiops-master/build/CMakeFiles/CMakeError.log".```

0xfederama, 2020-06-13:
I found it. I had to `git clone https://github.com/PixlOne/logiops`.

closed 2020-06-13 by 0xfederama

PixlOne, 2020-06-15:
Yeah, downloading the repository as a zip doesn't get the submodule for hidpp. However, cloning the repo doesn't require this.