PixlOne/logiops #72: MX Master 3 not detected on USB unifying receiver until switching to BT and back

gshipley, 2020-06-13:
Love this driver - so thanks!

I am having a weird issue on pop_os 20.04 / Ubuntu 20.04 in that my MX Master 3 is not detected on the unifying receiver until I switch to Bluetooth mode and then back to Unifying receiver.

$ sudo logid
[INFO] Found Logitech USB Receiver on /dev/hidraw15 // keyboard (G915)
[INFO] Found Logitech USB Receiver on /dev/hidraw6 -// Mouse (MX Master 3)
[INFO] Link lost to // Switch mouse selection to slot #2 
[INFO] MX Master 3 detected: device 255 on /dev/hidraw19 // Mouse is detected and connected via BT 
[INFO] MX Master 3 detected: device 0 on /dev/hidraw19
[INFO] MX Master 3 (Device 0 on /dev/hidraw19) disconnected //switch back to selection 1
[INFO] MX Master 3 (Device 255 on /dev/hidraw19) disconnected
[INFO] MX Master 3 detected: device 3 on /dev/hidraw6 //mouse is now detected using unifying receiver

Not a huge issue as I am just using logid on BT but thought this may be a stumbling block for others new to your package with a MX Master 3.

PixlOne, 2020-06-14:
Does this same behaviour occur if you turn your mouse off and back on again?

Ritte88, 2020-06-24:
I've the same mouse and issue, and yes if you toggle the power without changing to BT it helps with detection. 
It's the same thing with MX Keys.

Here's an interesting part of the info:
[INFO] Found Logitech USB Receiver on /dev/hidraw2
**[INFO] Link lost to** 

It's aware to that it did loose connection to MX Master 3 when turning of the power, this is before turning on the power where it's visible to logiops.

neacsugeorge, 2020-07-01:
Hi there, I have an MX Master 3 too, paired with the USB Receiver and I can confirm that once the service is started I have to toggle OFF/ON the mouse to have it detected by "logid". Afterwards, everything works.

System: Ubuntu 20.04 LTS
Hardware: Razer Blade 15

PixlOne, 2020-07-13:
Test out the refactor branch and see if that fixes this issue.