PixlOne/logiops #74: Unable to install

annatronci, 2020-06-14:
I apologize in advance if I ask basic questions but I am a beginner with Linux (Kubuntu) ... I have just purchased an Mx 3 and I am trying to use it. I tried to follow the instructions but I don't understand the issue of packages..Where they are to be installed and how ... Below I insert the workflow that I followed and certainly the packages are missing..Thanks in advance
Anna

Installation using the Zip (https://github.com/PixlOne/logiops.git)

INSTRUCTIONS REQUIRED TO INSTALL LOGIOPS:

$ apt install cmake
$ git clone git@github.com: PixlOne / logiops.git
$ cd logiops /
$ mkdir build
$ cd build
$ cmake ..
$ make
$ sudo make install

NECESSARY PACKAGES:
C ++ 14 compliler
cmake
libevedev
libudev
libconfig

anna @ anna-pc: ~ $ git clone https://github.com/PixlOne/logiops.git
Clone in 'logiops' in progress ...
remote: Enumerating objects: 563, done.
remote: Counting objects: 100% (563/563), done.
remote: Compressing objects: 100% (326/326), done.
remote: Total 563 (delta 323), reused 419 (delta 189), pack-reused 0
Reception of items: 100% (563/563), 109.68 KiB | 710.00 KiB / s, done.
Delta resolution: 100% (323/323), done.
anna @ anna-pc: ~ $ cd logiops /
anna @ anna-pc: ~ / logiops $ mkdir build
anna @ anna-pc: ~ / logiops $ cd build
anna @ anna-pc: ~ / logiops / build $ cmake ..

PixlOne, 2020-06-15:
I don't use any Ubuntu-based distros so I don't exactly know if this is correct but to install the dependencies, you should just need to run `sudo apt install cmake libevdev-dev libudev-dev libconfig++-dev`. Then you should be able to run the build commands in the readme.

I've been getting some issues like this and as a result, I'm planning to add install instructions for each distro. Can you test this out and get back to me? I want to make sure it works before I add this to the readme.

Thanks.

annatronci, 2020-06-15:
anna@anna-pc:~$ sudo apt install cmake libevdev-dev libudev-dev libconfig++-dev
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
cmake è già alla versione più recente (3.16.3-1ubuntu1).
libconfig++-dev è già alla versione più recente (1.5-0.4build1).
libudev-dev è già alla versione più recente (245.4-4ubuntu3.1).
Pacchetti suggeriti:
  libevdev-doc
I seguenti pacchetti NUOVI saranno installati:
  libevdev-dev
0 aggiornati, 1 installati, 0 da rimuovere e 0 non aggiornati.
È necessario scaricare 48,6 kB di archivi.
Dopo quest'operazione, verranno occupati 286 kB di spazio su disco.
Continuare? [S/n] S
Scaricamento di:1 http://it.archive.ubuntu.com/ubuntu focal/main amd64 libevdev-dev amd64 1.9.0+dfsg-1 [48,6 kB]
Recuperati 48,6 kB in 0s (119 kB/s)     
Selezionato il pacchetto libevdev-dev:amd64 non precedentemente selezionato.
(Lettura del database... 385878 file e directory attualmente installati.)
Preparativi per estrarre .../libevdev-dev_1.9.0+dfsg-1_amd64.deb...
Estrazione di libevdev-dev:amd64 (1.9.0+dfsg-1)...
Configurazione di libevdev-dev:amd64 (1.9.0+dfsg-1)...
Elaborazione dei trigger per man-db (2.9.1-1)...
anna@anna-pc:~$ cd logiops/
anna@anna-pc:~/logiops$ mkdir build
mkdir: impossibile creare la directory "build": File già esistente
anna@anna-pc:~/logiops$ cd build
anna@anna-pc:~/logiops/build$ cmake ..
-- Checking for module 'libevdev'
--   Found libevdev, version 1.9.0
-- Checking for module 'libconfig'
--   Found libconfig, version 1.5
Could not find libhidpp include dir, getting submodule
-- systemd units will be installed at /lib/systemd/system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/anna/logiops/build
anna@anna-pc:~/logiops/build$ make
Scanning dependencies of target hidpp
[  1%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/misc/Log.cpp.o
[  2%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/misc/CRC.cpp.o
[  4%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/RawDevice_linux.cpp.o
[  5%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/DeviceMonitor_linux.cpp.o
[  7%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hid/UsageStrings.cpp.o
[  8%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/Dispatcher.cpp.o
[ 10%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/SimpleDispatcher.cpp.o
[ 11%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/DispatcherThread.cpp.o
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In constructor ‘HIDPP::DispatcherThread::DispatcherThread(const char*)’:
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:70:8: warning: suggest explicit braces to avoid ambiguous ‘else’ [-Wdangling-else]
   70 |     if (!checkReportDescriptor (rdesc))
      |        ^
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In member function ‘virtual void HIDPP::DispatcherThread::sendCommandWithoutResponse(HIDPP::Report&)’:
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:108:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  108 |     if (forceLongReports() && report.type() == Report::Short)
      |     ^~
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:110:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  110 |  _dev.writeReport (report.rawReport ());
      |  ^~~~
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp: In member function ‘virtual std::unique_ptr<HIDPP::Dispatcher::AsyncReport> HIDPP::DispatcherThread::sendCommand(HIDPP::Report&&)’:
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:115:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  115 |     if (forceLongReports() && report.type() == Report::Short)
      |     ^~
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:117:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  117 |  std::unique_lock<std::mutex> lock (_command_mutex);
      |  ^~~
/home/anna/logiops/src/logid/hidpp/src/libhidpp/hidpp/DispatcherThread.cpp:121:77: warning: missing initializer for member ‘HIDPP::DispatcherThread::Command::response’ [-Wmissing-field-initializers]
  121 |  auto it = _commands.insert (_commands.end (), Command { std::move (report) });
      |                                                                             ^
[ 13%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/Device.cpp.o
[ 14%] Building CXX object src/logid/hidpp/src/libhidpp/CMakeFiles/hidpp.dir/hidpp/Report.cpp.o
[... 173 lines cut ...]
-- Installing: /usr/include/hidpp/hidpp20/ProfileFormat.h
-- Installing: /usr/include/hidpp/hidpp20/UnsupportedFeature.h
-- Installing: /usr/include/hidpp/hidpp20/IBatteryLevelStatus.h
-- Installing: /usr/include/hidpp/hidpp20/IMouseButtonSpy.h
-- Installing: /usr/include/hidpp/hidpp20/IFeatureSet.h
-- Installing: /usr/include/hidpp/hidpp20/IOnboardProfiles.h
-- Installing: /usr/include/hidpp/hidpp20/IReprogControls.h
-- Installing: /usr/include/hidpp/hidpp20/ISmartShift.h
[... cut ...]
-- Installing: /usr/include/hidpp/hidpp20/IAdjustableDPI.h
-- Installing: /usr/include/hidpp/hidpp20/IRoot.h
-- Installing: /usr/include/hidpp/hidpp20/Error.h
-- Installing: /usr/include/hidpp/hidpp20/Device.h
-- Installing: /usr/include/hidpp/hidpp20/MemoryMapping.h
-- Installing: /usr/include/hidpp/hidpp20/IWirelessDeviceStatus.h
-- Installing: /usr/include/hidpp/hidpp20/IHiresScroll.h
-- Installing: /usr/include/hidpp/hidpp20/IReset.h
-- Installing: /usr/lib/libhidpp.so
anna@anna-pc:~/logiops/build$ ^C
anna@anna-pc:~/logiops/build$

PixlOne, 2020-06-15:
So I'm assuming that worked. I'll add that to the readme, thanks.

annatronci, 2020-06-15:
No doesn`t work ..the run doesn`t start.....

PixlOne, 2020-06-15:
Can you post the log of  `sudo logid -v`?

annatronci, 2020-06-15:
anna@anna-pc:~$ sudo logid -v
[sudo] password di anna: 
[ERROR] I/O Error while reading /etc/logid.cfg: FileIOException
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)

annatronci, 2020-06-15:
this is my config file : 
devices: ( 
{ 
    name: "MX Master 3"; 
    smartshift: 
    { 
        on: false; 
        threshold: 30; 
    }; 
    hiresscroll: 
    { 
        hires: false; 
        invert: false; 
        target: false; 
    } ; 
    dpi: 1000; 

    buttons: ( 
        { 
            # back button 
            cid: 0x53; 
            action = 
            { 
                type: "Keypress"; 
                keys: ["KEY_LEFTALT","KEY_LEFT"]; 
            }; 
        }, 
        { 
            # forward button
            cid: 0x56; 
            action = 
            { 
                type: "Keypress"; 
                keys: ["KEY_LEFTALT","KEY_RIGHT"]; 
            }; 
        }, 
        { 
            # gesture button 
            cid: 0xc3; 
            action = 
            { 
                type: "Gestures"; 
                gestures: ( 
                    { 
                        direction: "Up"; 
                        mode: "OnRelease"; 
                        action = 
                        { 
                            type: "Keypress";
                            keys: ["KEY_LEFTALT","KEY_LEFTCTRL","KEY_UP"]; 
                        }; 
                    }, 
                    { 
                        direction: "Down"; 
                        mode: "OnRelease"; 
                        action = 
                        { 
                            type: "Keypress"; 
                            keys: ["KEY_LEFTALT"," KEY_LEFTCTRL","KEY_DOWN"]; 
                        }; 
                    }, 
                    { 
                        direction: "Left"; 
                        mode: "OnRelease";
                        action = 
                        {
                            type: "Keypress"; 
                            keys: ["KEY_LEFTSHIFT","KEY_LEFTALT","KEY_LEFTCTRL","KEY_DOWN"]; 
                        }; 
                    }, 
                    { 
                        direction: "Right"; 
                        mode: "OnRelease"; 
                        action = 
                        { 
                            type: "Keypress "; 
                            keys: ["KEY_LEFTSHIFT","KEY_LEFTALT","KEY_LEFTCTRL","KEY_UP"]; 
                        } 
                    },
                    { 
                        direction: "None"; 
                        mode: "OnRelease";
                        action = 
                        { 
                            type: "Keypress"; 
                            keys: ["KEY_LEFTCTRL","KEY_W"]; 
                        } 
                    } 
                ); 
            }; 
        }, 
        { 
            # SmartShift button 
            cid: 0xc4; 
            action = 
            { 
                type = "ToggleSmartshift"; 
            }; 
        } 
    ) ; 
} 
);

annatronci, 2020-06-15:
anna@anna-pc:~$ sudo logid -v
[sudo] password di anna: 
[ERROR] I/O Error while reading /etc/logid.cfg: FileIOException
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[... 333 lines cut ...]
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
^A[DEBUG] Receiver on /dev/hidraw0: Device 4 paired
[DEBUG] Receiver on /dev/hidraw0: Device 5 paired
[INFO] Connection established to MX Master 3
[INFO] Connection established to MX Keys
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[DEBUG] Receiver on /dev/hidraw0: Device 4 paired
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[INFO] Connection established to MX Keys
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[... cut ...]
[DEBUG] Time out on default device: /dev/hidraw4 (possibly asleep)
[DEBUG] Receiver on /dev/hidraw0: Device 5 paired
[DEBUG] Receiver on /dev/hidraw0: Device 4 paired
[INFO] Connection established to MX Master 3
[DEBUG] Time out on default device: /dev/hidraw5 (possibly asleep)
[INFO] Connection established to MX Keys
[DEBUG] Time out on default device: /dev/hidraw2 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw6 (possibly asleep)
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)

annatronci, 2020-06-15:
..i tried using even the sstandard config file but i doesn`t start....

PixlOne, 2020-06-16:
Are you sure you put the default config file in `/etc/logid.cfg`?

annatronci, 2020-06-16:
There`s no logid.config file in etc...what i have to do?

PixlOne, 2020-06-16:
You're supposed to copy logid.example.cfg from the repo (as root) to `/etc/logid.cfg`. After that, you can tune the config file to your liking.

annatronci, 2020-06-16:
anna@anna-pc:~$ sudo cp /home/anna/Scaricati/logid.example.cfg /etc/
[sudo] password di anna: 
anna@anna-pc:~$ sudo systemctl start logid
anna@anna-pc:~$ sudo
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p prompt] [-T timeout] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p prompt] [-T timeout] [-u user] file ...
anna@anna-pc:~$ 

##Now what happens?##

PixlOne, 2020-07-15:
All you need to do now is get the latest version and [follow the wiki to learn how to configure it](https://github.com/PixlOne/logiops/wiki/Configuration).

closed 2020-07-15 by PixlOne