PixlOne/logiops #75: Disabled mouse movement permanently

Euronames, 2020-06-14:
Hey, first of all, thank you so much for making this, it is really something that I have been looking for for a long time :) 

So, I got a little excited to try this on my MX Master, so I started by copying the config from your [example](https://github.com/PixlOne/logiops/blob/master/logid.example.cfg). I tried using the gestures, but then the cursor froze and is not working anymore... 

Wiping the config and restarting logid with systemctl does not fix the problem. 

Is there any way to fix this or reset the mouse completely? 

Cheers :+1:

Euronames, 2020-06-14:
Optput of  `sudo logid -v ` is : 

```
[INFO] MX Master detected: device 255 on /dev/hidraw6
[DEBUG] Available CID: 0x50
[DEBUG] Available CID: 0x51
[DEBUG] Available CID: 0x52
[DEBUG] Available CID: 0x53
[DEBUG] Available CID: 0x56
[DEBUG] Available CID: 0xc3
[DEBUG] Available CID: 0xc4
[DEBUG] Available CID: 0xd7
[DEBUG] 0x800 was pressed but no action was found.
[INFO] MX Master detected: device 0 on /dev/hidraw6
[DEBUG] Available CID: 0x50
[DEBUG] Available CID: 0x51
[DEBUG] Available CID: 0x52
[DEBUG] Available CID: 0x53
[DEBUG] Available CID: 0x56
[DEBUG] Available CID: 0xc3
[DEBUG] Available CID: 0xc4
[DEBUG] Available CID: 0xd7
[DEBUG] 0x800 was pressed but no action was found.

```

PixlOne, 2020-06-15:
What Linux distribution and kernel version are you running (run `uname -r` and post the output)?

This seems to work fine on my machine on Arch Linux 5.6.15-arch1-1.

Euronames, 2020-06-15:
Output of `uname -r` is: 
5.4.44-1-MANJARO

It seems like the mouse movement is completely disabled on the mouse itself. I have tried connecting it to other machines (both Windows and Mac) and the issue still persists  🤷‍♂️

PixlOne, 2020-06-15:
Wow, that is a really strange issue.

Have you tried restarting the device? I'm guessing you have already and that hasn't worked. Also does this occur over unifying and bluetooth?

If all else fails, I've recently discovered a way to completely reset the mouse to its out-of-box state but it's fairly complex.

Euronames, 2020-06-15:
You are right, restarting did nothing.  I am using Bluetooth since I lost the unifier 😅... 

I have a sneaking suspicion that it may be hardware related and that it was a coincidence that it happens just as I tried Logiops out... I will try [this](https://youtu.be/7ZDHFJEgyiM) fix, and return with the results!  

Should it not be hardware related, I may have to try your big red reset button 🙃

Euronames, 2020-07-06:
Sorry for the late response... It totally was a hardware issue and the video fixed it. Will close issue!

closed 2020-07-06 by Euronames